adafruit/circuitpython #5734: Please add collections deque to CircuitPython

jftheoret, 2021-12-16:
Hello!

Please activate collections deque in CircuitPython. Thanks!

tannewt, 2021-12-16:
What board are you using? Some of them won't have space for it.

jftheoret, 2021-12-16:
I'm using the Raspberry Pi Pico.

I was actually looking in the code to see if I could activate it and
recompile?

Thanks :)

tannewt, 2021-12-16:
You can set `MICROPY_PY_COLLECTIONS_DEQUE` to `(1)` to enable it.

jftheoret, 2021-12-16:
Thanks!!

I will definitely try it. It's also going to be an opportunity to learn how
to build it :)

jftheoret, 2021-12-17:
Another quick question for you - I have changed the code, recompiled and
re-flashed the firmware.

The import now works.

When I try to use it:

self.stored_chars = deque([], 4)

I get a ValueError. I am trying to create an empty queue with a maximum of
4 elements.

Can you please give me a hint as how I can investigate this further, as all
I have is a ValueError without any other trace information?

*code.py output:Traceback (most recent call last):  File "code.py", line
130, in <module>  File "code.py", line 73, in __init__ValueError: Code done
running.*

Thanks!!

dhalbert, 2021-12-17:
Try `deque((), 4)`. The MicroPython code does not support initialization, and only allows an empty tuple, not an empty list.

jftheoret, 2021-12-17:
Excellent, thank you!

polygnomial, 2022-04-17:
I built pycubed_v05 (it's an M4 board with SAMD51J20), with `MICROPY_PY_COLLECTIONS_DEQUE = 1`, but the module is not available in CircuitPython. The `.o` file is missing from the `build/py` folder. I also tried with a couple of other combinations of `MICROPY_PY_COLLECTIONS` etc, but have not gotten it to work. Any suggestions?

dhalbert, 2022-10-16:
Fjxed by #6474. Still omitted on some very small builds.

closed 2022-10-16 by dhalbert