adafruit/circuitpython #5739: Core code crashed hard

ErlendFj, 2021-12-17:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0 on 2021-09-20; Adafruit Feather M4 Express with samd51j19
```

### Code/REPL

```python
USB keept dropping out, plugged in battery and got error message Crash into the HardFault_Handler. Maybe this is a brownout related bug/behaviour?
```

### Behavior

cannot run any code

### Description

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
Please file an issue with the contents of your CIRCUITPY drive at

### Additional information

How do I get out of safe mode?

jerryneedell, 2021-12-17:
Press the Reset button or power cycle to get out of safe mode.

ErlendFj, 2021-12-17:
It drops right back in to safe mode. I will try to format the 'drive' if I can get USB to stay alive for long enough.

jerryneedell, 2021-12-17:
In safe mode, you can delete you code.py or boot.py if they are causing the problem. You can also erase the file system in safe mode.

dhalbert, 2021-12-17:
This may be because your code is using lots of NeoPixels or other power-hungry hardware, and the USB power is sagging.

What else is connected besides the Feather M4 Express?

ErlendFj, 2021-12-17:
Did it, I'm now back to Hello World. I suspect that as I debugged my original code, more and more hardware got powered and I gradually moved into brown-out territory. Will reinforce power supply and try again. Thanks for helping.

dhalbert, 2021-12-17:
I'm closing this for now, but feel free to discuss this kind of support issue on discord: https://adafru.it/discord or the forums: https://forums.adafruit.com/viewforum.php?f=60.

It is also possible you are exercising a bug that is causing the hard fault. In that case, reopen this or open a new issue, with the specific code that is causing that problem. Thanks.

closed 2021-12-17 by dhalbert