adafruit/circuitpython #5743: Time methods report "No long integer support" on BCM

RetiredWizard, 2021-12-17:
### CircuitPython version

```python
Adafruit CircuitPython 7.1.0-beta.3 on 2021-12-13; Raspberry Pi Zero 2W with rp3a0
```

### Code/REPL

```python
import time
time.time()
time.localtime()
time.mktime()
```

### Behavior

```Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NotImplementedError: No long integer support

RetiredWizard, 2021-12-17:
I'm assuming (hoping :) ) the broadcom raspberry pi platform isn't space constrained so long integer support shouldn't be disabled?

tannewt, 2021-12-20:
Yup, this should just be a flag we need to enable.

RetiredWizard, 2021-12-23:
I'm not sure if this is the correct approach, but I stole the following code from the raspberrypi port and placed it at the top of ports/broadcom/mpconfigport.mk which seems to have enabled long integer support.

         # Define an equivalent for MICROPY_LONGINT_IMPL, to pass to $(MPY-TOOL) in py/mkrules.mk
         # $(MPY-TOOL) needs to know what kind of longint to use (if any) to freeze long integers.
         # This should correspond to the MICROPY_LONGINT_IMPL definition in mpconfigport.h.

         ifeq ($(LONGINT_IMPL),NONE)
         MPY_TOOL_LONGINT_IMPL = -mlongint-impl=none
         endif

         ifeq ($(LONGINT_IMPL),MPZ)
         MPY_TOOL_LONGINT_IMPL = -mlongint-impl=mpz
         endif

         ifeq ($(LONGINT_IMPL),LONGLONG)
         MPY_TOOL_LONGINT_IMPL = -mlongint-impl=longlong
         endif

         # All raspberrypi ports have longints.
         LONGINT_IMPL = MPZ

I can now evaluate a timestamp (i.e. time.localtime(33333333333)), however  most of the time function calls now return the following:

         Traceback (most recent call last):
           File "<stdin>", line 1, in <module>
         RuntimeError: RTC is not supported on this board

tannewt, 2021-12-23:
That looks right to me. Want to make a PR? The BCM chips don't have an RTC officially. We could probably fake it with the system timer though. I think it has microsecond resolution.

RetiredWizard, 2021-12-23:
Okay, PR for the long integers submitted :). I don't know if you want to link this issue and open a new one for the RTC or just keep this one going.

tannewt, 2021-12-28:
I think it's best to have a new issue for RTC.

RetiredWizard, 2021-12-28:
Okay, I'll go ahead and close this one and open a new one for the RTC Support

closed 2021-12-28 by RetiredWizard