adafruit/circuitpython #5745: Adafruit ESP32-S2 Feather hard crashes with wifi.radio.ping

BlueBackbite, 2021-12-17:
### CircuitPython version

```python
Adafruit CircuitPython 7.1.0-beta.3 on 2021-12-13; Adafruit Feather ESP32S2 with ESP32S2
Board ID:adafruit_feather_esp32s2
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2019 ladyada for Adafruit Industries
# SPDX-FileContributor: Modified by Reppad
#
# SPDX-License-Identifier: MIT

import os
import board
import neopixel
import ipaddress
import wifi
import rtc
import time
import random
import socketpool

# Get wifi details and more from a secrets.py file
try:
    from secrets import secrets
except ImportError:
    print("WiFi secrets are kept in secrets.py, please add them there!")
    raise

try:
    import json as json_module
except ImportError:
    import ujson as json_module

# Track time we've been unable to ping certain IPs
time_modem_ping = time.time()
# VERIFY this is the default modem management IP, it ussually is, and *ussually* responds to ping
modemIP = ipaddress.ip_address("192.168.100.1")

time_external_ping = time.time()
external_ips = [ipaddress.ip_address("1.1.1.1"),ipaddress.ip_address("8.8.8.8"),ipaddress.ip_address("8.8.4.4"),ipaddress.ip_address("1.0.0.1"),ipaddress.ip_address("208.67.222.222"),ipaddress.ip_address("208.67.220.220")]

# Delay the number of attempts we make, so we're not spamming ping all the time.
time_delay = time.time()

# Track when last power cycle happened.
time_reset = time.time()

#how often in seconds to check for connectivity (Will ping every x seconds) Don't flood the network with pings!
check_interval = 1

# count the number of pings
ping_counter = 0

#Set our hostname? (I think this works)
wifi.radio.hostname = "ModPowerMon"

print("Connect wifi")
wifi.radio.connect(secrets['ssid'],secrets['password'])
HOST = repr(wifi.radio.ipv4_address)
GATEWAY = repr(wifi.radio.ipv4_gateway)
DNS = repr(wifi.radio.ipv4_dns)
MYNAME = repr(wifi.radio.hostname)
PORT = 80        # Port to listen on
print(HOST,PORT)

# Use below for Most Boards
status_light = neopixel.NeoPixel(
    board.NEOPIXEL, 1, brightness=0.2
) 

# Run checks to see if we can reach outside the network...
def check_connectivity():
    global time_reset, time_modem_ping, time_external_ping, ping_counter
    sing_ip = random.choice(external_ips)
    print("pinging ",sing_ip)
    ping_result = wifi.radio.ping(sing_ip)
    if ping_result is None:
        #unable to ping site, try modem
        ping_modem_result = wifi.radio.ping(modemIP)
        if ping_modem_result is None and time.time() - time_modem_ping > 3600:
            # Toggle our latching relay to cut power to the modem for 2 minutes
            print("Haven't been able to reach any outside networks, for more than an hour")
            time_reset = time.time()
            time_external_ping = time.time()
            time_modem_ping = time.time()
        else:
            time_modem_ping = time.time()
    else:
        ping_counter += 1
        time_external_ping = time.time()
        ping_modem_result = wifi.radio.ping(modemIP)
        if ping_modem_result is None:
            print("Was able to ping external IP but not modem, that's weird.")
        else:
            ping_counter += 1
            time_modem_ping = time.time()
            print("Was able to ping external IP and modem, should be good to go!",ping_counter)

print(f"hostname: {MYNAME} gateway IP: {GATEWAY} DNS: {DNS}")

while True:
    # Our main loop where we have the server poll for incoming requests
    # Could do any other background tasks here, like reading sensors
    if time.time() > time_external_ping + check_interval and time.time() > time_delay + check_interval:
        time_delay = time.time()
        check_connectivity()
```

### Behavior

After 30 pings, the board hard faults into safe mode.
The same behavior happens when using larger time intervals between pings.

### Additional information

Nothing else is attached to the board.

anecdata, 2021-12-17:
I've reproduced your result.

But this ran a few hundred loops without issue:
```py
import supervisor
import time
import random
import wifi
import socketpool
import ipaddress
from secrets import secrets

SERVERS = ("1.1.1.1", "1.0.0.1",
           "1.1.1.2", "1.0.0.2",
           "8.8.8.8", "8.8.4.4",
           "9.9.9.9", "149.112.112.112",
           "208.67.220.220", "208.67.222.222",
           "208.67.220.123", "208.67.222.123")

print("Connecting...")
wifi.radio.connect(secrets['ssid'],secrets['password'])
pool = socketpool.SocketPool(wifi.radio)

loop = 0
while True:
    ipv4 = wifi.radio.ipv4_gateway
    result = wifi.radio.ping(ipv4)
    print(f'LAN {loop:>3} {supervisor.ticks_ms():>9} {str(ipv4):15}     {result}s')

    port = random.choice((53, 80, 443))
    ipv4 = ipaddress.ip_address(pool.getaddrinfo(random.choice(SERVERS), port)[0][4][0])
    result = wifi.radio.ping(ipv4)
    print(f'WAN {loop:>3} {supervisor.ticks_ms():>9} {str(ipv4):15} {port:>3} {result}s')
    print()
    loop += 1
    time.sleep(1)
```

I'm not sure at this point what would account for the difference.

BlueBackbite, 2021-12-17:
I would think it's not ping then, perhaps something with the global variables being set in-loop or issues with time?

If I remove the global variables and just have the function return True or False, it crashes in 2 loops or 4 pings.

I just got into python. I have no clue.

anecdata, 2021-12-18:
Can you try to see what minimal example still produces the hard fault? Get rid of everything not essential to the demo.

BlueBackbite, 2021-12-18:
It hard faults with this code very fast
```Python
import ipaddress
import wifi

# Get wifi details and more from a secrets.py file
try:
    from secrets import secrets
except ImportError:
    print("WiFi secrets are kept in secrets.py, please add them there!")
    raise

modemIP = ipaddress.ip_address("192.168.100.1")
# count the number of pings
ping_counter = 0

print("Connect wifi")
wifi.radio.connect(secrets['ssid'],secrets['password'])

def check_connectivity():
    ping_result = wifi.radio.ping(modemIP)
    if ping_result is None:
        print("ping modem fail")
    else:
        print("Was able to ping modem, should be good to go!")

while True:
    ping_counter += 1
    print("loops",ping_counter)
    check_connectivity()
```
    
But if you remove the function completely, it simply hangs:
```Python
import ipaddress
import wifi

# Get wifi details and more from a secrets.py file
try:
    from secrets import secrets
except ImportError:
    print("WiFi secrets are kept in secrets.py, please add them there!")
    raise

modemIP = ipaddress.ip_address("192.168.100.1")
# count the number of pings
ping_counter = 0

print("Connect wifi")
wifi.radio.connect(secrets['ssid'],secrets['password'])

while True:
    ping_counter += 1
    print("loops",ping_counter)
    ping_result = wifi.radio.ping(modemIP)
    if ping_result is None:
        print("ping modem fail")
    else:
        print("Was able to ping modem, should be good to go!")
```

dhalbert, 2022-02-14:
Closing in favor of #5745 (dupe).

closed 2022-02-14 by dhalbert

anecdata, 2022-02-14:
Sorry, forgot about this one, I should have searched first.