adafruit/circuitpython #5763: Read the docs link for CircuitPython don't work

dannystaple, 2021-12-22:
### CircuitPython version

```python
NA
```

### Code/REPL

```python
NA
```

### Behavior

NA

### Description

I get a 404 if I click the Edit On Github links in the Documentation. This may be a readthedocs issue, or a configuration issue in the doc generation.

Reproducing this:

- Open a document in CircuitPython readthedocs - ie https://circuitpython.readthedocs.io/en/latest/shared-bindings/rp2pio/#rp2pio.StateMachine.readinto. 
- Click the "Edit on GitHub" button.

Result:

- Takes you to the link "https://github.com/adafruit/circuitpython/blob/main/shared-bindings/rp2pio/index.rst" - which is a 404 on github.

Expected:
It should probably link to https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/bindings/rp2pio - since the module doc is in the C code here.

jepler, 2022-01-08:
Because of the way we generate the documentation, it may be best if we can just turn off those links. it's probably possible in sphinx configuration, if someone wants to go looking for it.

tekktrik, 2022-01-14:
Here's how to do it: https://docs.readthedocs.io/en/stable/guides/remove-edit-buttons.html

tekktrik, 2022-01-14:
I can give this a shot

tekktrik, 2022-01-15:
Not entirely sure this fixed the issue, I still see the links in the top right corner...

jepler, 2022-01-15:
[image]
 yeah I am seeing the same thing, unfortunately.

jepler, 2022-01-15:
I confirmed that github built main after #5863 was merged in

tekktrik, 2022-01-15:
I'll see if I can build RTD locally and follow up to this with a solution.  I've had issues building locally before but I'll see if I can over come that and solve this!

PR #6301 (2022-04-25): Add docs/templates to templates in conf.py
Fixes #5763 with the fix suggested by @neradoc by adding `docs/templates` to `conf.py`.  Building locally, it removes the "View page source" so this should be the fix!
closed 2022-04-25 by dhalbert