adafruit/circuitpython #5768: Broadcom port doesn't include io module

RetiredWizard, 2021-12-23:
### CircuitPython version

```python
Adafruit CircuitPython 7.1.0-beta.3 on 2021-12-13; Raspberry Pi Zero 2W with rp3a0
Adafruit CircuitPython 7.1.0-beta.3-174-gbea052695-dirty on 2021-12-22; Raspberry Pi Zero 2W with bcm2837
```

### Code/REPL

```python
import io
```

### Behavior

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'io'

tannewt, 2021-12-23:
Want to try enabling this as well? We should probably set FULL_BUILD=1 and then disable the things we don't support. That'd catch a lot of stuff like this.

RetiredWizard, 2021-12-23:
I can test and put a PR in with FULL_BUILD enabled easily enough, but I don't think I could figure out everything that might need to be disabled.

I think I can track down the flag needed to enable just IO though if you want me to dig into that instead.

Unless of course you were talking to your team and not me :D

jerryneedell, 2021-12-23:
i'm puzzled -- what is "io" I don't see a model names "io" in the support matrix.

RetiredWizard, 2021-12-23:
[Input Output streams](https://circuitpython.readthedocs.io/en/latest/docs/library/io.html)

dhalbert, 2021-12-23:
It is `MICROPY_PY_IO`, so it doesn't get found properly with the other native modules.

jerryneedell, 2021-12-23:
sorry -- now I see it in the rp2040 build

jerryneedell, 2021-12-23:
> It is `MICROPY_PY_IO`, so it doesn't get found properly with the other native modules.

ah so it is not in the matrix --  thanks  -- it does show up with help('modules')

jerryneedell, 2021-12-23:
I can try to enable FULL_BUILD and work on the disables -- but probably not until next week

RetiredWizard, 2021-12-23:
Whew :D, I just tried to compile with FULL_BUILD and realized I'm way out of my depth.... Identifying the MICRO_PY_IO flag solves my short term needs anyway. Thanks!!!

PR #5819 (2022-01-11): Add Piunora and Zero. Also, enable full build
Fixes #5768 and fixes #5631
closed 2022-01-11 by dhalbert