adafruit/circuitpython #5770: How can I make the circuitpython work with the bootloader(*nosd.hex) which don't include the softdevice?

bigbearishappy, 2021-12-23:
I am working with nrf52840 board now. And I can make the circuitpython work fine with the adafruit bootloader which included softdevice.
However, when I change to the bootloader which don't included the softdevice.The circuitpython is not working.

I tried to modify the ports/nrf/bluetooth/bluetooth_common.mk to remove the "CFLAGS += -DSOFTDEVICE_PRESENT". I also tried to 
change the ld file to delete the flash region of softdevice.
But I still can't make it work.

Is there any suggestion or plan to make it?
Thanks~

dhalbert, 2021-12-23:
We have never tested without the SoftDevice, so there may be errors in the logic, and also the linking map may be wrong, since it assumes the firmware is placed after the SoftDevice in flash.

The attraction of using CircuitPython on the nRF boards is that it has BLE support, so we have always assumed the SoftDevice would be present. There are other chip families without BLE that we would recommend otherwise.

bigbearishappy, 2021-12-24:
> We have never tested without the SoftDevice, so there may be errors in the logic, and also the linking map may be wrong, since it assumes the firmware is placed after the SoftDevice in flash.
>
> The [...]

Thank you for your reply.
It seems that we need to do more work to make the circuitpython work fine with the nosd bootloader.
By the way, I find that arduino nano33 use the nrf52840 too. And it's bootloader don't include something like the softdevice.
Maybe I can have a try to make circuitpython work by referring to the nano33 ble.

closed 2021-12-24 by bigbearishappy

dhalbert, 2021-12-24:
The Arduino Nano33 uses mbed and a completely different BLE software stack, which doesn't use the SoftDevice.