adafruit/circuitpython #5772: Update UF2 Family for nRF52833 boards

jpconstantineau, 2021-12-23:
[nRF52 bootloader project](https://github.com/adafruit/Adafruit_nRF52_Bootloader) has updated the family for the nRF52833 to 0x621E937A for the nRF52833.

This will ensure that 840 firmware won't be able to be flashed to a 833 and vice-versa.

We need to update the Makefile [here](https://github.com/adafruit/circuitpython/blob/main/ports/nrf/Makefile#L267) with some if statement to check for the nRf52833.

I don't know if there is anything else that needs updating.

tannewt, 2021-12-23:
Here is a similar change done for ESP: https://github.com/adafruit/circuitpython/pull/5771

jpconstantineau, 2021-12-23:
Changes look relatively simple. Is there a variable that indicates the target? (For the nrf port)

closed 2021-12-26 by microDev1