adafruit/circuitpython #5777: How to get out of the Safe Mode?

bujna94, 2021-12-24:
### CircuitPython version

```python
7.0
```

### Code/REPL

```python
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Crikey!
Crash into the HardFault_Handler.
Please file an issue with the contents of your CIRCUITPY drive at
https://github.com/adafruit/circuitpython/issues

Press any key to enter the REPL. Use CTRL-D to reload.
```

### Behavior

need to get out of the safe mode

jerryneedell, 2021-12-24:
Press the RESET button or power cycle to exit safe mode.

microDev1, 2021-12-24:
You might need to clean your `code.py` before resetting or else you would end up in safe mode again on reset.

closed 2021-12-24 by microDev1