adafruit/circuitpython #5787: croadcome - Pin.c setting of JTAG -- should it have pull-down?

jerryneedell, 2021-12-28:
When the Pins are reset in
https://github.com/adafruit/circuitpython/blob/main/ports/broadcom/common-hal/microcontroller/Pin.c#L53
for the JTAG pins 22-27, the Pins it set to ALT4 then it returns without setting a PullDOWN.
Looikng ar the Datasheet, these pins are listed as having pull-downs.

Sould the be set here?

It seem sot be working as it and it and I f set them as Pull-down it still works. I'm just curious if they should be set .

here is the change I made to test it.
```
diff --git a/ports/broadcom/common-hal/microcontroller/Pin.c b/ports/broadcom/common-hal/microcontroller/Pin.c
index 1cf86f2b3..f3355a7be 100644
--- a/ports/broadcom/common-hal/microcontroller/Pin.c
+++ b/ports/broadcom/common-hal/microcontroller/Pin.c
@@ -48,14 +48,15 @@ void reset_pin_number(uint8_t pin_number) {
     pin_in_use[pin_number] = false;
     never_reset_pin[pin_number] = false;
     // Reset JTAG pins back to JTAG.
+    BP_PULL_Enum pull = BP_PULL_NONE;
     if (22 <= pin_number && pin_number <= 27) {
         gpio_set_function(pin_number, GPIO_FUNCTION_ALT4);
+        pull = BP_PULL_DOWN;
         return;
     } else {
         gpio_set_function(pin_number, GPIO_FUNCTION_INPUT);
     }
     // Set the pull to match the datasheet.
-    BP_PULL_Enum pull = BP_PULL_NONE;
     if (pin_number < 9 ||
         (33 < pin_number && pin_number < 37) ||
         pin_number > 45) {
```

tannewt, 2021-12-28:
I looked at the current state of pins prior to the first reset and they didn't all match the datasheet. So, I'd go by what works rather than what the datasheet says.

jerryneedell, 2021-12-28:
Sounds good. Thanks.

jerryneedell, 2021-12-28:
Actually, one more question...
If you set it to ALT4  and return but don't explicitly set the Pull (up/down) then I guess it just stays as it was. Is that OK or do you want top be explicitly setting it to something?

tannewt, 2021-12-29:
Ya, we should probably reset it to the state we want. That's good to know about reading pull back. I think I'm doing that now.

jerryneedell, 2021-12-29:
> Ya, we should probably reset it to the state we want. That's good to know about reading pull back. I think I'm doing that now.

Would you like me to submit a PR for the change I made above to set them to pull-down or since you are likely making other changes do you want to do that yourself?

tannewt, 2021-12-30:
Go ahead @jerryneedell. I'm working on Zero support.

tannewt, 2022-01-06:
This is done in #5798. Thanks @jerryneedell

closed 2022-01-06 by tannewt