adafruit/circuitpython #5792: Vectorio Ignores Palette colors (pi zero 2W)

rsbohn, 2021-12-29:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0-alpha.0 on 2021-12-03; Raspberry Pi Zero 2W with rp3a0
```

### Code/REPL

```python
import board
import displayio
import vectorio

display=board.DISPLAY
splash=displayio.Group()
# background?
palette=displayio.Palette(1)
palette[0]=0xFF00FF
circle = vectorio.Circle(radius=24,x=display.width//2, y=display.height//2, pixel_shader=palette)
display.show(splash)
splash.append(circle)
```

### Behavior

You see a black circle on a black background.

It may help to add a non-black background bitmap.

```
px=displayio.Palette(1)
px[0]=0x666699
bg=displayio.Bitmap(display.width,display.height,1)
tbg=displayio.TileGrid(bg, pixel_shader=px)
splash.append(tbg)
```

### Description

Expected result: You see a magenta circle on a black background.

### Additional information

This was working in Adafruit CircuitPython 7.2.0-alpha.0-127-gcad8c1fcd on 2021-12-15; Raspberry Pi Zero 2W with bcm2837

https://github.com/adafruit/circuitpython/pull/5728

rsbohn, 2021-12-29:
This is what happens when you keep old kernels around. Sorry.

closed 2021-12-29 by rsbohn

dglaude, 2022-01-08:
Dear @rsbohn I know this ticket is closed, but I did keep a link to it as an example of graphics on Pi Zero 2W.
I am testing with `adafruit-circuitpython-raspberrypi_zero2w-en_US-20220107-fb6ab72.disk.img` (latest).

Your code is working in the REPL, but does nothing visible (it stay in "text mode" when I run that from code.py).
And I have `print("something")` for debug, and my (your) code is running.
I was wondering if you had any clue on what I am doing wrong.

Thanks in advance.

dglaude, 2022-01-09:
You can ignore the above, after a night of sleep I added:
```
import time

time.sleep(10)
```

And the graphic is visible 10 seconds.

rsbohn, 2022-01-12:
Sorry I didn't see this earlier. Glad that you were able to get it working.