adafruit/circuitpython #5805: Pink Adafruit Feather RP2040 Board ID Issue?

tdawry, 2022-01-01:
I am not sure if this is the right place for this, as I am new to working with hardware (sorry!).

The [_Pink Adafruit Feather RP2040 - Free for orders $99 or more Product ID: 5299_](https://www.adafruit.com/product/5299) may have the incorrect _Board ID:_ as a default value.

- If _not_ updating, and connecting to the device via Thonny, the _Board ID:_ suggests it is a _Raspberry Pi Pico_ of no particular variant, and it is missing a number of libraries (like _board_). Connecting via a serial connection does work, but the missing libraries persist. On Windows 10 it does not show up as _CIRCUITPY_ (or anything else). The bootloader does show in Explorer if booting to it.

- Following the CircuitPython instructions on the Adafruit website leads to updating the firmware with a file such as _adafruit-circuitpython-adafruit_feather_rp2040-en_US-7.1.0.uf2_ but this fails. It presents errors such as _"ERROR thonny.plugins.micropython.backend: Crash in backend"_. Connecting via a serial connection does not seem to work or present an error here.

- Updating the _Board ID:_ to _adafruit_feather_rp2040_ and only then updating to the newest CircuitPython, _adafruit-circuitpython-adafruit_feather_rp2040-en_US-7.2.0-alpha.1.uf2_, seems to fix the main issues here. The board is still not accessible in Thonny but does show up in Explorer, and will allow for a _code.py_ to be copied over and executed.

PC settings with which this was tested:
_Edition	Windows 10 Education
Version	21H1
Installed on	‎7/‎6/‎2020
OS build	19043.1415
Experience	Windows Feature Experience Pack 120.2212.3920.0_

jerryneedell, 2022-01-01:
I loaded circuitpython-adafruit_feather_rp2040-en_US-7.2.0-alpha.1.uf2 to mine then set Thonny for CircuitPython(generic) and selected the RP2040 port -- in Configure Interpreter.  It seem to  be working OK.

Note: as delivered, CircuitPython is not installed. You do have to install it.

tdawry, 2022-01-01:
It may be something on my end or something particular to this board. I am able to get it working but if I return to the bootloader and try to start from there I get the _Crash in backend_ error referenced above and it does not show as a drive in my file explorer unless I am running the bootloader.

Changing the line in _INFO_UF2.TXT_ from _Board-ID: RPI-RP2_ to _Bourd_ID: adafruit_feather_rp2040_ and then dragging the *.uf2 over does reboot it and then everything seems to work. Thonny with CircuitPython(generic) does work here, too.

If this is not reproducible, it can be closed. Hopefully if someone else runs into this problem this shows up in a search result! Thanks for the help.

dhalbert, 2022-01-01:
I am confused. The `INFO_UF2.TXT` file is a fake file that shows up when you are in bootloader mode, in the RPI-RP2 drive, which is not a real drive. It is not possible to edit this file. All boards using the RP2040 chip show this:
```py
UF2 Bootloader v2.0
Model: Raspberry Pi RP2
Board-ID: RPI-RP2
```
Are you changing the `BOOT_OUT.TXT` file instead?

tdawry, 2022-01-02:
I may have been, I cannot seem to reproduce it now that it is up and running as I do not know how to return the board back to stock (firmware yes, but not the whole thing).

This is probably a BCAK error, sorry about that!

closed 2022-01-02 by tdawry