adafruit/circuitpython #5806: Float formatting shortcomings/errors

dastels, 2022-01-01:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0 on 2021-09-20; Adafruit Grand Central M4 Express with samd51p20
```

### Code/REPL

```python
>>> print("{0:,.2f}".format(283525.34561))
283525.26
>>> value = 283525.34561
>>> print(f"{value:,.2f}")
283525.26
```

### Behavior

In both cases the output should be "283,525.35"

### Description

Comma specifier is ignored, decmail part rounding is incorrect.

dhalbert, 2022-01-01:
The 30-bit floating point format used in CircuitPython only allows about 5-1/2 digits of accuracy. Try the number here and turn off the two lowest bits: https://www.h-schmidt.net/FloatConverter/IEEE754.html. The non-implementation of `,` is a missing feature.

snkYmkrct, 2024-10-09:
Comma issue duplicate of https://github.com/adafruit/circuitpython/issues/4776 ?

closed 2024-10-09 by tannewt