adafruit/circuitpython #5814: Compiled the Seeed_XIAO_nRF52840_Sense board is wrong start address

0hotpotman0, 2022-01-06:
Hi I compile the Seeed_XIAO_nRF52840_Sense board is work on my local, coz I add the SOFTDEV_VERSION=7.0.1 in behind the "make BOARD=...SOFTDEV_VERSION=7.0.1",          I think I downloaded that board from the Circuitpython web one does not add "SOFTDEV_VERSION=7.0.1", It cause the start address wrong.

[image]

I think the system build the uf2 is the default make BOARD=...,   how can I make it to add SOFTDEV_VERSION=7.0.1 behind the make BOARD=...

How can I solve this case, please give me some advice, thank you.

0hotpotman0, 2022-01-06:
If build the board without the "SOFTDEV_VERSION=7.0.1", the start address will be 0x26000

tannewt, 2022-01-06:
Have you tried setting `SOFTDEV_VERSION=7.0.1` in `ports/nrf/boards/Seeed_XIAO_nRF52840_Sense/mpconfigboard.mk`? I think that should override the setting from `mpconfigport.mk`.

0hotpotman0, 2022-01-07:
Hi @tannewt   
Where can I add "SOFTDEV_VERSION=7.0.1", 

Can I add behind the "USB_PRODUCT = "Seeed XIAO nRF52840 Sense"
[image]

tannewt, 2022-01-07:
I think it can go anywhere in that file. It's included by the `Makefile` before the port version.

dhalbert, 2022-03-09:
Closing this for now. Please reopen if you are still having trouble.

closed 2022-03-09 by dhalbert