adafruit/circuitpython #5818: Chat using open source platform?

kj7rrv, 2022-01-07:
Hello,
I saw that the circuit python chat is on Discord. Are there any rooms on open source platforms like Matrix bridged to the Discord server? I'm interested in joining but I won't use a proprietary app.

dhalbert, 2022-01-07:
We have not done any bridging. You can use Discord from a web browser, so there is no app that you need to install.

kj7rrv, 2022-01-07:
Okay. Are there any plans to bridge it in the future?

dhalbert, 2022-01-07:
No plans, but if you want to work on it, you could. We have had no other requests for Matrix. I would like to understand your motivation and whether the discord browser app is acceptable to you, and if not, why not.

kj7rrv, 2022-01-07:
Okay. I just want to avoid using any closed source software (including Web apps) that I don't have to.

kj7rrv, 2022-01-07:
Approximately how high is the message volume in the chat?

dhalbert, 2022-01-07:
It's quite high volume. Are you willing to take a look through the web app? Our discord server has >23000 members and about 30 active channels.

You can ask questions in our forum too: https://forums.adafruit.com/viewforum.php?f=60. That has a closed source back-end.

kj7rrv, 2022-01-07:
Okay. I'm okay with a closed source backend as long as I don't have to run closed source code on my device.

closed 2022-01-07 by dhalbert

kj7rrv, 2022-01-07:
I won't be able to set it up right away, but would you be okay with adding a bridge bot running on my server?

dhalbert, 2022-01-07:
Are you willing to discuss this _on_ discord for now? Our GitHub issues are not really appropriate for this. You can bring it up in #help-with-community. Our server is well-moderated, and we don't know how well the mdoeration would interact with the bridge (e.g., would deleted messages be deleted over the bridge, banned users, etc.?)

kj7rrv, 2022-01-07:
Deleted messages should be deleted over the bridge. I don't think there's a way to ban users over the bridge.
Yeah, that's okay. I'll join the discord for now.

kj7rrv, 2022-01-07:
I'm ony Android phone right now so I had to install the app. Where there should be a box to type in, it just shows a button saying resend verification email. I've tapped it several times and haven't received a verification email.

dhalbert, 2022-01-07:
Sounds like a spam-catching issue. Maybe wait until you are back at a desktop.

kj7rrv, 2022-01-07:
Would you be willing to discuss it in the forums instead?

dhalbert, 2022-01-07:
Discord is better -- the other moderators are there. Please try to resolve your discord registration issues on a desktop when you get a chance. If you fail then open a forum thread. I saw you join as @kj7rrv and have pinged you in the #help-with-community channel. Please don't use GitHub as a chat server. Dozens of people get an email every time there's a post in this issue.