adafruit/circuitpython #5823: keypad reset does not get pressed keys

Neradoc, 2022-01-08:
The description of  keypad.KeyMatrix.reset(self) states: 
> Any key that is already pressed at the time of this call will therefore immediately cause a new key-pressed event to occur.

However it seems that's not the case, and instead you have to wait for the time interval to pass.
I ran this code on latest on a raspberry pico with 7.1.0 with a wire connecting the pins.
> Adafruit CircuitPython 7.2.0-alpha.1-54-gfb6ab7239 on 2022-01-07; Raspberry Pi Pico with rp2040

```py
import board
import keypad
import time

keys = keypad.KeyMatrix(
    row_pins=(board.GP6,),
    column_pins=(board.GP3,),
    columns_to_anodes=False
)

keys.reset()
event = keys.events.get()
print(event)

keys.reset()
time.sleep(0.001)
event = keys.events.get()
print(event)

keys.reset()
time.sleep(0.01)
event = keys.events.get()
print(event)

keys.reset()
time.sleep(0.1)
event = keys.events.get()
print(event)
```
```py
boot.py output:
None
None
None
<Event: key_number 0 pressed>
```
Is the description wrong or unclear, or is it a bug ?

dhalbert, 2022-01-13:
In this case, "immediately" means it will be noticed in the first cycle of scanning. I was trying to emphasize that you _will_ get an event as soon as possible, instead of it being ignored.

jepler, 2022-01-13:
would it be reasonable to complete a scan "in the foreground" before returning from the constructor? It is nice if the keypad can sense keys 'immediately', to avoid having to loop or wait in boot.py.

dhalbert, 2022-01-13:
> would it be reasonable to complete a scan "in the foreground" before returning from the constructor? It is nice if the keypad can sense keys 'immediately', to avoid having to loop or wait in boot.py.
The constructor could simply delay for the scan time, maybe plus a little bit, since the scanning will be enabled at the bottom of the constructor. The same could be done for `reset()`.

Neradoc, 2022-04-15:
Closed by #6248
Tested with the code above on latest.
```py
boot.py output:
<Event: key_number 0 pressed>
<Event: key_number 0 pressed>
<Event: key_number 0 pressed>
<Event: key_number 0 pressed>
```

closed 2022-04-15 by Neradoc