adafruit/circuitpython #5824: ESP32-S3-DevKitC-1-N8R8

prplz, 2022-01-09:
The `espressif_esp32s3_devkitc_1` board definition does not work with the N8R8 devkit because it uses octal psram. I was able to get it working by coping the sdkconfig and some never_reset pins from `espressif_esp32s3_box` board.

I opened an issue instead of a PR because there are decisions to be made about how to name and manage these different variants (more discussion at #5678).

tannewt, 2022-01-10:
I think our naming should just follow the N16R8V style suffixes.

tannewt, 2022-01-13:
The V is flash voltage. We shouldn't need a different build for it.

[image]

closed 2022-01-13 by tannewt

prplz, 2022-01-13:
> The V is flash voltage. We shouldn't need a different build for it.

It would still need `CIRCUITPY_ESP_FLASH_SIZE` though right?

Also if [this Amazon page](https://www.amazon.com/Espressif-ESP32-S3-DevKitC-1-N16R8V-Development-Board/dp/B09MFC7D23) is correct, it has octal flash, I'm not sure if that requires some more config changes.

tannewt, 2022-01-14:
Ya, we'd need flash size because it controls partition layout.

The [datasheet for the modules](https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-1_wroom-1u_datasheet_en.pdf) says it's only QSPI.

PR #5848 (2022-01-13): Esp32s3 devkits
Fixes #5824

Renames the existing boards to their part name for consistency and adds N8R8 variant with octal ram.
Tested on N8R8, it passes basic neopixel blinking tests and is able to allocate a 7MB array.