adafruit/circuitpython #5825: Please add degree (Sign °) in terminalio.FONT

CrackXT, 2022-01-09:
The standard font (terminalio.FONT) does not contain the degree (°) sign (?) And thus it is ignored. Interestingly, there is no error if you "define" it in the code as a symbol to be displayed.

Example line of code:
cpu_label.text = 'CPU Temp. [R]: {: .1f} ° C'.format (microcontroller.cpu.temperature)

Is then output / shown on the display as follows:
CPU Temp. [R]: 38.7C

As far as ok except for the ° is not displayed and the C slips right away.

If I replace the terminalio.Font with another font, the degree symbol is displayed, but it doesn't make sense to me why such an international standard symbol is missing and you have to switch to a separate font if CircuitPython already contains one.

Therefore the request to add the degree symbol in the terminalio.FONT, thank you.

Kind regards 
CrackXT

deshipu, 2022-01-09:
The `terminalio.FONT` uses the font that is built in into the CircuitPython firmware itself. For space limits reasons, that font only includes the characters that are used in the internal messages in CircuitPython itself, because its main use is to display those messages.

You can load the same font from a file, and then all the characters will be included. The font used by default is `ter-u12n.bdf`.

tannewt, 2022-01-10:
I understand why you are suggesting this @CrackXT but as @deshipu points out, we don't have room in most builds for this.

closed 2022-01-10 by tannewt

CrackXT, 2022-01-11:
Thanks @deshipu for your feedback. However, I cannot understand why I should include the already integrated font again using additional lines of code, just so that more characters are enabled (?) There. That takes away unnecessary storage space because it actually exists twice ... or I didn't understand it ;(.

With your reference to which font it is in CP, I looked "deeper" into the code and found it here
https://github.com/adafruit/circuitpython/blob/main/tools/fonts/ter-u12n.bdf

If you look in there then the degree symbol is stored in line 2156?! What now makes me guess that it is already integrated, but for whatever reason it is not displayed ... Bug? The character is also available with Fontforge open.

Thanks @tannewt for your feedback. Will the font be reduced in number of characters for each board beforehand?

Kind regards
CrackXT

tannewt, 2022-01-11:
> Thanks @tannewt for your feedback. Will the font be reduced in number of characters for each board beforehand?

We do the filtering here: https://github.com/adafruit/circuitpython/blob/main/tools/gen_display_resources.py#L50-L101

CrackXT, 2022-01-12:
Thanks @tannewt for your feedback. Ok, if I understand that correctly then the code line in its link reduces the font to "ASCII-128" and then decodes to UTF-8.
visible_ascii = bytes (range (0x20, 0x7F)). decode ("utf-8")

It's kind of confused (and for me inconsistent) that the REPL then apparently gets its characters for display from somewhere else ... and is therefore actually not compliant with the code.

Example line of code:
print ('CPU Temperature [R]: {: .1f} ° C'.format (microcontroller.cpu.temperature))

Is then output / shown in REPL as follows:
CPU Temperature [R]: 35.2 ° C

I suspect that "ASCII-256" aka Extended ASCII is not a future option or is not implemented for other reasons (space)?
visible_ascii = bytes (range (0x20, 0xFF)). decode ("utf-8")

Thank you anyway, at least that's how I found out about the CP font and no longer need to tinker with one that I created myself.

Kind regards
CrackXT

tannewt, 2022-01-12:
> It's kind of confused (and for me inconsistent) that the REPL then apparently gets its characters for display from somewhere else ... and is therefore actually not compliant with the code.

What do you mean? I'm not sure what you are pointing out.

> I suspect that "ASCII-256" aka Extended ASCII is not a future option or is not implemented for other reasons (space)?

We could bring in more characters into builds that have a lot of flash space. However, we tend to run out of room on most platforms over time.

CrackXT, 2022-01-14:
Thanks @tannewt for your feedback. I mean, if only the ASCII-128 font is loaded, why are characters displayed in the REPL that the code in the µC has not "read" at all and therefore does not transmit.

This difference is confusing for me, I would expect that the character is also missing in the REPL.

Kind regards
CrackXT

dhalbert, 2022-01-14:
Do you mean the REPL on a built-in display, or the REPL when you are connected via a serial terminal (Mu or `screen`, etc.)?

CrackXT, 2022-01-14:
Thanks @dhalbert for your feedback. The REPL when i connected via a serial terminal  (i use Mu).

dhalbert, 2022-01-14:
@CrackXT When you use Mu, the internal font is not used. You are using a font on the host computer, which does have the degree sign. The missing degree sign is only an issue when you are looking at the REPL on an attached display.

CrackXT, 2022-01-21:
Thanks @dhalbert for your feedback.

lynt-smitka, 2023-08-22:
I ran into the same problem when writing some tutorials.

In order to display the degree character in builtin font, I wrote a small wrapper that tries to load the character from an external birmap font when the character is not found in builtin.

Then I extracted the degree glyph in FontForge from https://github.com/adafruit/circuitpython/blob/70ce576390c7b21007b1f75f16e6fc591c0f61d7/tools/fonts/ter-u12n.bdf and created tiny bdf file contains only that glyph.

```
import terminalio
from adafruit_bitmap_font import bitmap_font

class ExtraFont:

    def __init__(self):
        self.builtin_font = terminalio.FONT
        self.extra_font = bitmap_font.load_font("/extra.bdf")

    def get_glyph(self, code_point):
        glyph = self.builtin_font.get_glyph(code_point)
        if glyph is None:
            glyph = self.extra_font.get_glyph(code_point)
        return glyph

    def get_bounding_box(self):
        return self.builtin_font.get_bounding_box()
```

You can see an example and the bdf file here: https://github.com/MakerClassCZ/Picopad/tree/CircuitPython/circuitpython/sensors/intro/level3