adafruit/circuitpython #5828: Error in str.format for non number formatting

TG-Techie, 2022-01-10:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0-alpha.1-58-g211dc53d4-dirty on 2022-01-10; TG-Watch with nRF52840
```

### Code/REPL

```python
'{:09}'.format('Jan')
```

### Behavior

The behavior  circuitpython is:
```python
>>> '{:09}'.format('Jan')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: '=' alignment not allowed in string format specifier
```

### Description

the behavior on python 3.10 is:
```python
>>> '{:09}'.format('Jan')
'Jan000000'
```

It took a good bit to figure out what the error was, I’m not sure if the behavior with this error is intentional. 
The error message was decently misleading while debugging my code. (because I didn’t have any `=`s in the text I was using)

### Additional information

Taking a quick look at the source, it looks like that error message is commented as for not numbers but is also the default case... which doesn't necessarily have an = in it
https://github.com/adafruit/circuitpython/blob/476b17b9876b6003b550c15f652ef7560dffa309/py/objstr.c#L1378-L1389

I'm more than glad to PR a different error message! I’m not sure if this is a large enough edge case to merit a PR?

Or PR a check for an equal and display this message and display a more generic "we don’t know what to do with the given formatter" otherwise.

jepler, 2022-01-10:
Since this is about the core, I've prepared a fix and offered it to micropython. We can make a decision whether to take it ourselves, independently of whether they accept the change.

jepler, 2022-01-10:
Python folks considered it a bug fix, as far as I can tell: 
 * https://github.com/python/cpython/pull/11270

dhalbert, 2023-10-28:
This is now merged in and is fixed in 9.0.0, at least:
```py
>>> '{:09}'.format('Jan')
'Jan000000'
```

closed 2023-10-28 by dhalbert