adafruit/circuitpython #5833: ulab documentation mentions nonexistent `ulab.spectrum`

Thynix, 2022-01-11:
At least as of `Adafruit CircuitPython 7.1.0`, places like [`ulab.numpy.fft`](https://circuitpython.readthedocs.io/en/latest/shared-bindings/ulab/numpy/fft/index.html) documentation refer to a nonexistent "spectrum" function:

> See also ~ulab.extras.spectrum, which computes the magnitude of the fft, rather than separately returning its real and imaginary parts.

I haven't been able to figure out whether it's supposed to exist, or its removal was intentional. 6.3 had the similarly-named [spectrogram](https://circuitpython.readthedocs.io/en/6.3.x/shared-bindings/ulab/fft/index.html#ulab.fft.spectrogram), which has since been removed.

tannewt, 2022-01-11:
@v923z and @jepler

v923z, 2022-01-11:
Indeed, this is a very or relic, and the reference should be replaced by `scipy.signal.spectrogram`. I'll take care of it.

tekktrik, 2022-04-20:
Not sure how to PR over to what seems to be the commit CircuitPython is using, but I submitted https://github.com/v923z/micropython-ulab/pull/522 to the base repo for this.

tekktrik, 2022-04-22:
Okay, the fix should be merged into master for 8.0.0!

dhalbert, 2022-10-23:
Fixed.

closed 2022-10-23 by dhalbert