adafruit/circuitpython #5839: New version of CircuitPython 7.1.0 and associated new libraries seems to have broken deep sleep on MagTag

wavesailor, 2022-01-11:
### CircuitPython version

```python
Adafruit CircuitPython 7.1.0 on 2021-12-28; Adafruit MagTag with ESP32S2
Board ID:adafruit_magtag_2.9_grayscale
```

### Code/REPL

```python
import time
import terminalio
import displayio
import adafruit_imageload
from adafruit_display_text import label
from adafruit_magtag.magtag import MagTag
from secrets import secrets

# --| USER CONFIG |--------------------------
METRIC = False  # set to True for metric units
# -------------------------------------------

# ----------------------------
# Define various assets
# ----------------------------
BACKGROUND_BMP = "/bmps/weather_bg.bmp"
ICONS_LARGE_FILE = "/bmps/weather_icons_70px.bmp"
ICONS_SMALL_FILE = "/bmps/weather_icons_20px.bmp"
ICON_MAP = ("01", "02", "03", "04", "09", "10", "11", "13", "50")
DAYS = ("Monday", "Tuesday", "Wednesday", "Thursday", "Friday", "Saturday", "Sunday")
MONTHS = (
    "January",
    "February",
    "March",
    "April",
    "May",
    "June",
    "July",
    "August",
    "September",
    "October",
    "November",
    "December",
)
magtag = MagTag()

# ----------------------------
# Backgrounnd bitmap
# ----------------------------
magtag.graphics.set_background(BACKGROUND_BMP)

# ----------------------------
# Weather icons sprite sheet
# ----------------------------
icons_large_bmp, icons_large_pal = adafruit_imageload.load(ICONS_LARGE_FILE)
icons_small_bmp, icons_small_pal = adafruit_imageload.load(ICONS_SMALL_FILE)

# /////////////////////////////////////////////////////////////////////////

def get_data_source_url(api="onecall", location=None):
    """Build and return the URL for the OpenWeather API."""
    if api.upper() == "FORECAST5":
        URL = "https://api.openweathermap.org/data/2.5/forecast?"
        URL += "q=" + location
    elif api.upper() == "ONECALL":
        URL = "https://api.openweathermap.org/data/2.5/onecall?exclude=minutely,hourly,alerts"
        URL += "&lat={}".format(location[0])
        URL += "&lon={}".format(location[1])
    else:
        raise ValueError("Unknown API type: " + api)
[... 207 lines cut ...]
This all worked perfectly before I upgraded to the latest version of CircuitPython 7.1.0 and  associated new libraries  
```
adafruit-circuitpython-adafruit_magtag_2.9_grayscale-en_US-7.1.0.uf2
adafruit-circuitpython-bundle-7.x-mpy-20220105.zip
```

### Additional information

These are the libraries I have installed on the MagTag
```
[adafruit_bitmap_font]   
[adafruit_display_shapes]
[adafruit_display_text]
[adafruit_imageload]
[adafruit_io]
[adafruit_magtag]
[adafruit_portalbase]
adafruit_datetime.mpy 
adafruit_fakerequests.mpy
adafruit_miniqr.mpy  
adafruit_pixelbuf.mpy 
adafruit_requests.mpy 
neopixel.mpy
simpleio.mpy
```

DavePutz, 2022-01-12:
I tested this using both CP 7.0.0 and 7.1.0; monitoring with a USB power monitor. In both cases, once the script goes into deep sleep I see about 200uA of current draw. I was not able to test using a battery.

dhalbert, 2022-01-13:
I also tested the program above, using 7.0.0 and 7.1.0, and got about 230uA during sleeping. I then also tested using an extremely simple deep sleep example, from https://learn.adafruit.com/deep-sleep-with-circuitpython/alarms-and-sleep#timealarm-deep-sleep-3078700-7. Again, the sleep current was about 230uA. This value is typical, and is what we first saw when deep sleep was first introduced in CircuitPython, in 6.2.0.

I am testing with a Nordic PPK2 power monitor, which simulates a battery. When I am testing, there is nothing plugged in to the USB port.

I am not sure why you are seeing increased battery consumption. @wavesailor Are you powering this through the battery connector, with nothing plugged in to the USB port, or are you using the USB port for power?

wavesailor, 2022-01-14:
This is strange now - I went back and reloaded the code from the Weather App in code.py and reset it.
It seems to run fine now with the original code from the Weather App.

I have my own version where I added and changed a few things which is where I first discovered the problem. So I'm loading my code into code.py and testing it out. I'll give feedback over the next few days.

PS. What do you use to measure the current draw?

dhalbert, 2022-01-14:
> PS. What do you use to measure the current draw?

I am using this :https://www.adafruit.com/product/5048. That's what I used for measurements and the screenshots in https://learn.adafruit.com/deep-sleep-with-circuitpython/power-consumption.

patja, 2022-01-23:
I did some testing with a MagTag project using CircuitPython 7.1.1 and did not see this problem.  I put an Extech 380942 clamp ammeter on the positive battery wire and saw current draw bounce around 80 - 120 mA while running, and then drop to about 2 mA while in deep sleep.  I ran through several sleep/wake cycles and it never varied.

dhalbert, 2022-01-23:
I am going to close this for now. @wavesailor If you can reproduce, please let us know and reopen.

dhalbert, 2022-01-23:
@patja Thank you for testing!

closed 2022-01-23 by dhalbert

patja, 2022-01-24:
@wavesailor I noticed your code shows a different method of going into deep sleep than I had seen documented.  Is there a reason you are not using the alarm module as suggested here:  https://learn.adafruit.com/deep-sleep-with-circuitpython/alarms-and-sleep

Just a suggestion, but maybe try using the alarm module and see if the behavior changes.

dhalbert, 2022-01-24:
`magtag.exit_and_deep_sleep()` is a convenience routine that ultimately calls the `alarm` module `exit_and_deep_sleep()`. I just reviewed the library code and it looks OK.

wavesailor, 2022-03-21:
I redid Circuit python and libraries from scratch. This seemed to have solved my issue.
Thanks