adafruit/circuitpython #5843: ESP32S2-TFT SPI conflict with pin 36

jerryneedell, 2022-01-12:
### CircuitPython version

```python
7.2.0-alpha.1
```

### Code/REPL

```python
Adafruit CircuitPython 7.2.0-alpha.1 on 2021-12-29; Adafruit Feather ESP32-S2 TFT with ESP32S2
>>> import board
>>> spi = board.SPI()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: D36 in use
>>>
```

### Behavior

see above

### Additional information

Thsi works OK on 7.1.0

ladyada, 2022-01-12:
this is that thing with the stuff @jepler 😓  :D

jerryneedell, 2022-01-12:
Fixed by #5846 
using bus.SPI(...) still repost SCK in use
but board.SPI() works

jepler, 2022-01-12:
`bus.SPI(...)` is expected not to work, because the pins are in use by the TFT.  I think it's a bug in 7.1.x that it seemed to work.

`board.SPI()` is (with #5846) the exact same SPI object used by the display, so they can both use it without conflicting over the pins.

PR #5846 (2022-01-13): feather esp32s2 tft: fix sharing of main display bus
Closes: #5843
closed 2022-01-13 by dhalbert