adafruit/circuitpython #5847: rotaryio pin behaviour

oliwetti, 2022-01-12:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0-alpha.1-74-g9c6e34df0 on 2022-01-11; Raspberry Pi Pico with rp2040
```

### Code/REPL

```python
# uses circuitpython
from time import sleep
import board
import rotaryio

enc = rotaryio.IncrementalEncoder(board.GP10, board.GP11, divisor=1)
last_position = enc.position
while True:
    if(enc.position != last_position):
        print(enc.position)
        last_position = enc.position
        sleep(0.01)
```

### Behavior

Nothing happens, no count, unusual pin behaviour at pin_a and pin_b

### Description

I used a normal alps encoder connected with a and b switch to 3V3 and got no counts so I tested the pin lever using a scope and a voltage or ampere meter:

no program: pins are low
Behaviour with no pulldown attached:
     If start the program the pins go to high (all other unused pins stay low)
Behaviour with 24kOhm pulldown
     If start the program the pins at aprox. 1,25 V (so somewhere in between)

I checked the rotaryio behavior against the countio behavior: With countio the pins stay low.

Maybe the rotaryio module switches the pullup on? I tested the rotary pins with an ampere meter and i got 56 uA PIN to GND
If I wire the encore to GND everything works fine.

I read the documentation, but I did not find, that the pullups are enabled at the pi pico when rotaryio is used

So one way would be to document this behaviour
Another way would be to turn off the internal pullups and leave it to the user
a third way would be to add a pull-option

tannewt, 2022-01-14:
Yup, I believe we do pull up automatically and agree it should be documented. The comment here should be updated so ReadTheDocs is updated: https://github.com/adafruit/circuitpython/blob/main/shared-bindings/rotaryio/IncrementalEncoder.c#L40

dhalbert, 2022-01-15:
Yes, we assume the encoder grounds its signals on transitions, and we set pull-ups accordingly in all the implementations.

pree-T, 2022-01-16:
May I work on it? Can someone help me as I am new to open source contribution?

dhalbert, 2022-01-16:
@pree-T Sure! See these guides for how to get started:
https://learn.adafruit.com/building-circuitpython 
https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github

Currently the only thing to do is to document that `rotaryio.IncrementalEncoder` assumes that the encoder's pins are connected to ground, and sets pullups on the pins accordingly. The documentation is inline in the C code in  [`circuitpython/shared-bindings/rotaryio/IncrementalEncoder.c`](https://github.com/adafruit/circuitpython/blob/main/shared-bindings/rotaryio/IncrementalEncoder.c). The actual implementations are in `circuitpython/ports/*/common-hal/rotaryio/...`

muranyia, 2022-02-20:
From a learner's perspective, there are two points missing from the documentation:

1. pull is set automatically
2. direction is set automatically

I was unsure until I found this issue, so you might want to include 2. as well.

heygauri, 2023-10-02:
Hello @dhalbert, I am looking forward to contributing to this issue. The following changes are proposed after reading your previous comment and @muranyia's comment.

sumitra@sumitra:~/opensource/circuitpython$ git diff
diff --git a/shared-bindings/rotaryio/IncrementalEncoder.c b/shared-bindings/rotaryio/IncrementalEncoder.c
index 24df2b2d1..669b85c5f 100644
--- a/shared-bindings/rotaryio/IncrementalEncoder.c
+++ b/shared-bindings/rotaryio/IncrementalEncoder.c
@@ -35,7 +35,9 @@
 #include "shared-bindings/util.h"
 
 //| class IncrementalEncoder:
-//|     """IncrementalEncoder determines the relative rotational position based on two series of pulses."""
+//|     **"""IncrementalEncoder determines the relative rotational position based on two series of pulses. 
+//|       The rotaryio.IncrementalEncoder assumes that the encoder's pins are connected to ground and 
+//|       sets pull-ups and direction on the pins accordingly."""**
 //|
 //|     def __init__(
 //|         self, pin_a: microcontroller.Pin, pin_b: microcontroller.Pin, divisor: int = 4

PR #8453 (2023-10-02): Update documentation for rotaryio.IncrementalEncoder
Update the documentation in circuitpython/shared-bindings/rotaryio/IncrementalEncoder.c to explicitly state that rotaryio.IncrementalEncoder assumes the encoder's pins are connected to ground and sets pull-ups on the pins accordingly.

Closes #5847
commit fd4a6123 (2023-10-02): Update documentation for rotaryio.IncrementalEncoder

Update the documentation in circuitpython/shared-bindings/rotaryio/IncrementalEncoder.c
to explicitly state that rotaryio.IncrementalEncoder assumes the encoder's pins are
connected to ground and sets pull-ups on the pins accordingly.

Closes #5847
closed 2023-10-02 by tannewt