adafruit/circuitpython #5861: Bluetooth workflow: Can't connect to nRF52840

rsbohn, 2022-01-14:
### CircuitPython version

```python
Adafruit CircuitPython 7.1.0 on 2021-12-28; Adafruit Feather Bluefruit Sense with nRF52840
```

### Code/REPL

```python
n/a
```

### Behavior

When I try to "Request Bluetooth Device" from Chrome (code.circuitpython.org) 
I pick the 'CIRC' device and click 'Pair', then I get a Bluetooth busy spinner that never goes away.

### Description

Chrome 96.0.4664.110 (Official Build) (64-bit)
--enable-experimental-web-platform-features 
--enable-features=WebBluetoothNewPermissionsBackend
Windows 10 Version 21H1

### Additional information

Requesting any Bluetooth device...
script.js:520 > Requested CIRPY
script.js:374 connected BluetoothRemoteGATTServer {device: BluetoothDevice, connected: true}
script.js:382 (2) [BluetoothRemoteGATTService, BluetoothRemoteGATTService]
script.js:384 Getting Transfer Service...
ble-file-transfer.js:79 caught connection error DOMException: Authentication canceled. undefined
ble-file-transfer.js:100 caught write error TypeError: Cannot read properties of null (reading 'writeValueWithoutResponse')
    at FileTransferClient._write (ble-file-transfer.js:87)
    at FileTransferClient.listDir (ble-file-transfer.js:467)
    at async showBusy (script.js:186)
    at async fileExists (script.js:219)
    at async loadEditor (script.js:488)
    at async switchToDevice (script.js:392)
    at async onRequestBluetoothDeviceButtonClick (script.js:522)
    at async HTMLButtonElement.<anonymous> (script.js:598) TypeError: Cannot read properties of null (reading 'writeValueWithoutResponse')
    at FileTransferClient._write (https://code.circuitpython.org/_snowpack/pkg/@adafruit/ble-file-transfer.js:87:38)
    at FileTransferClient.listDir (https://code.circuitpython.org/_snowpack/pkg/@adafruit/ble-file-transfer.js:467:20)
    at async showBusy (https://code.circuitpython.org/assets/js/script.js:186:18)
    at async fileExists (https://code.circuitpython.org/assets/js/script.js:219:19)
    at async loadEditor (https://code.circuitpython.org/assets/js/script.js:488:9)
    at async switchToDevice (https://code.circuitpython.org/assets/js/script.js:392:5)
    at async onRequestBluetoothDeviceButtonClick (https://code.circuitpython.org/assets/js/script.js:522:9)
    at async HTMLButtonElement.<anonymous> (https://code.circuitpython.org/assets/js/script.js:598:9)
ble-file-transfer.js:100 caught write error TypeError: Cannot read properties of null (reading 'writeValueWithoutResponse')
    at FileTransferClient._write (ble-file-transfer.js:87)
    at FileTransferClient.listDir (ble-file-transfer.js:468)
    at async showBusy (script.js:186)
    at async fileExists (script.js:219)
    at async loadEditor (script.js:488)
    at async switchToDevice (script.js:392)
    at async onRequestBluetoothDeviceButtonClick (script.js:522)
    at async HTMLButtonElement.<anonymous> (script.js:598) TypeError: Cannot read properties of null (reading 'writeValueWithoutResponse')
    at FileTransferClient._write (https://code.circuitpython.org/_snowpack/pkg/@adafruit/ble-file-transfer.js:87:38)
    at FileTransferClient.listDir (https://code.circuitpython.org/_snowpack/pkg/@adafruit/ble-file-transfer.js:468:20)
    at async showBusy (https://code.circuitpython.org/assets/js/script.js:186:18)
    at async fileExists (https://code.circuitpython.org/assets/js/script.js:219:19)
    at async loadEditor (https://code.circuitpython.org/assets/js/script.js:488:9)
    at async switchToDevice (https://code.circuitpython.org/assets/js/script.js:392:5)
    at async onRequestBluetoothDeviceButtonClick (https://code.circuitpython.org/assets/js/script.js:522:9)
    at async HTMLButtonElement.<anonymous> (https://code.circuitpython.org/assets/js/script.js:598:9)
script.js:345 DOMException: Connection failed for unknown reason. undefined
script.js:525 Argh: disconnected
script.js:505 true
script.js:345 DOMException: Connection failed for unknown reason. undefined
script.js:505 true
script.js:345 DOMException: Connection failed for unknown reason. undefined
script.js:505 true

tannewt, 2022-01-18:
Does it work with older builds? If not, the web-editor repo is probably a better place for the issue: https://github.com/circuitpython/web-editor

rsbohn, 2022-01-19:
Tried with the 7.0.0 release, same result. Moving to web-editor.

closed 2022-01-19 by rsbohn