adafruit/circuitpython #5864: Hug Report for CircuitPython BareMetal on Raspberry Pi Zero W and Interrupt Step-Over

TheMindVirus, 2022-01-15:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0-alpha.1-95-g936c9b205n 2022-01-13;
Raspberry Pi Zero W with bcm2835;
BoardID:raspberrypi_zero_w
```

### Code/REPL

```python
print("Hello World!")
input(">>>")
```

### Behavior

REPL Lock-up with no errors or traceback, potentially ARM1176 Interrupt being called and returning to the same place in a loop.
The stack trace and other information matches that for the Pi Zero 2W: [Deep Dive w/Scott: Pi Zero 2W + CircuitPython #adafruit](https://www.youtube.com/watch?v=DHE9anKZrGw)

### Description

This happens whenever the normal procedure is followed (kernel7.img) for adding CircuitPython BareMetal to an existing installation of Raspberry Pi OS (and it has also been observed to do the same with flashing the disk.img.zip version).

As soon as you press enter over a serial terminal to write something in the REPL, it stops accepting user input and displays no error messages. Some research prior to seeing this suggests that interrupts are being triggered and are not returning to the correct place in memory after they complete. Please additionally see: https://github.com/themindvirus/pix-ies/tree/step-over for more.

The Pi Zero W, Pi Zero 2W, Pi 4 etc. have different generations of processor on board and also both have a different boot procedure and Interrupt Vector Table. The attached example is for the Pi 4 with other branches for Pi Zero, but I'm currently working on re-writing it for ARM1176 instead of Cortex-A72 or Cortex A53. Some assistance on creating an actual PR for this would be appreciated.

### Additional information

# OBS Studio REPL
[image]
# Pepper's Ghost REPL
[image]

TheMindVirus, 2022-01-15:
Additional required steps for Debian WSL (Windows Subsystem for Linux) include but are not limited to the following:
```
sudo apt update # Update Debian repositories list
apt-cache search # Find missing packages (information also available at Debian's websites)

sudo apt-get install build-essential git make # Base tools for development
# Additional cross-compilers such as arm-none-eabi-gcc may also be required

sudo apt-get install python3-pip # For installing python huffman module
sudo pip install huffman # For python imaging codecs
sudo apt-get install parted # For partitioning "kernel7.img"
sudo apt-get install dosfstools # For formatting with mkfs.fat
sudo apt-get install mtools # For copying MS-DOS files to Unix with mcopy
sudo apt-get install zip # For zip and unzip if not already installed

sudo git clone https://github.com/adafruit/circuitpython # Takes 10GB
cd circuitpython/ports/broadcom # Change directory into the workspace

sudo nano Makefile # and check BOARD=raspberrypi_zero_w
sudo nano peripherals/broadcom/vcmailbox.c # Comment line 115 that states:
# "#pragma GCC diagnostic ignored \"-Waddress-of-packed-member\""
sudo make V=2 # To start the build process

# If it complains about missing directories it may not have permission to make them without sudo or it might not have a required package
sudo mkdir mkfs.fat # vs. sudo apt-get install mtools

# The build output will be located in a folder matching the name of the board
cd build-raspberrypi_zero_w # The name of this folder may vary
ls # To list the folder contents

# Look for "firmware.disk.img.zip" and "firmware.kernel.img".
# See the steps in the learn guide for what to do with these build outputs.
# It involves either Extracting and Flashing the image with Raspberry Pi Imager
# or Renaming to "kernel7.img" and copying over to an existing SD installation.
```

From there we will need to find where interrupts are being handled with "eret" and make sure the "ELR" (Exception Link Register) returns to the correct place.
This is specific only to this instance of Broadcom chips which have this quirk.

TheMindVirus, 2022-01-15:
I'm closing this issue for now because it seems that the most recent main branch is working as expected.
However, there still remains the issue of it being updated on Amazon S3 which the Learn Guide is currently pointing to.

[image]

Both this and the CircuitPython build for the Adafruit Macropad RP2040 will help in creating an AVR8 port
for the ATMEGA16U2 on the Arduino Uno and USB2SERIAL in under 10KB with LUFA USB Mass Storage firmware.
Current Progress: https://github.com/TheMindVirus/micropython/tree/master/ports/avr8

closed 2022-01-15 by TheMindVirus

tannewt, 2022-01-18:
What link to S3 are you referring to?

Why are you porting MP to the AVR8 and ATMEGA16U2? Don't they have far too little RAM?

TheMindVirus, 2022-01-18:
> What link to S3 are you referring to?
> 
> Why are you porting MP to the AVR8 and ATMEGA16U2? Don't they have far too little RAM?

It may have all changed by now:
Your learn guide is here: https://learn.adafruit.com/circuitpython-on-raspberry-pi-bare-metal-no-os/installation
That takes you to circuitpython.org: https://circuitpython.org/board/raspberrypi_zero_w/
Which says "Browse S3": https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/raspberrypi_zero_w/en_GB/
...of which beforehand the CI had some updated builds, I tried one but it locked up at boot.
It seems that issue is resolved in the main branch if manually building from source as detailed above.

For AVR8 I'd like to see something additional other than just a direct 1:1 relay between Host PC and Microcontroller.
CircuitPython could eventually provide that flexibility to forward to multiple microcontrollers, even ones not originally physically connected to the "IoT Hub" over USB. The RAM is not so much of an issue so long as the build is optimised for size and 16KB.
That whole thing is just an experiment to see if it can be done, and it would be awesome to see Arduino and CircuitPython running on the same Uno.