adafruit/circuitpython #5869: sdcardio deinit hangs if spi.deinit() done first

jerryneedell, 2022-01-16:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0-alpha.1-109-gfce6d16c8 on 2022-01-15; Raspberry Pi Pico with rp2040
```

### Code/REPL

```python
import sdcardio
import busio
import digitalio
import board
import storage

# Connect to the card and mount the filesystem.
spi = busio.SPI(board.GP10, board.GP11, board.GP12)
cs = board.GP15
sdcard = sdcardio.SDCard(spi, cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")
spi.deinit()
print("spi dinited")
sdcard.deinit()
print("sdcard dinited")
```

### Behavior

The example code intentionally has the spi.deinit before the sd.deinit. Thsi may not make any sense, but I don't think it should hang.

```

Adafruit CircuitPython 7.2.0-alpha.1-109-gfce6d16c8 on 2022-01-15; Raspberry Pi Pico with rp2040
>>> import sdcardio_hang
spi dinited
```
It is hung  -- control-c will not interrupt

If the order if the deinits is reversed, then it works normally.

This came up when I ran the "maker Pi" demo and tried to do the SDCard test (button 3)
https://github.com/CytronTechnologies/MAKER-PI-PICO/blob/main/Out-of-the-box%20Demo%20Code/CircuitPython/code.py#L438

I think the demo used to work but it hangs now.

### Description

I wonder if this is due to this spi write done during the sdcardio deinit
https://github.com/adafruit/circuitpython/blob/main/shared-module/sdcardio/SDCard.c#L115
I'm not sure what the spi write will do if the spi has been deininted.

this was introduced recently (3 months ago) 

Again, It works fine if you "do the right thing" and deinit sdcardio first.
Is there some way to prevent the hang and throw an error if it is done wrong as in this demo?

jepler, 2022-01-16:
The new operation during deinit occurs to ensure that all data is written out to the card, so it's important.

The shared-bindings routines that wrap from Python to C like `busio_spi_configure` are where the 'check for deinit' logic is, so when sdcardio calls directly down to `common_hal_busio_spi_configure` it bypasses the usual check that leads to an exception, and instead does things like access hardware registers anyway.

It implies we need to move those checks to some new middle layer between shared-bindings and common-hal, or [less desirable] duplicate them.

jerryneedell, 2022-01-16:
I hesitated to file this issue because once I realized what was happening, it was so easy to avoid.
That said, it took me quite some time to figure out what was going on and why it was just hanging so I thought it best to document the issue. Perhaps the solution is "don't do that"!