adafruit/circuitpython #5870: Rebooting into bootloader doesn't work

jepler, 2022-01-16:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0-alpha.1-112-g9b32d549f-dirty on 2022-01-16; Adafruit Feather ESP32-S2 TFT with ESP32S2
```

### Code/REPL

```python
>>> import microcontroller
>>> microcontroller.on_next_reset(microcontroller.RunMode.BOOTLOADER)
>>> microcontroller.reset()
```

### Behavior

Board disconnects completely. (I thought it restarted as with a regular reset, but that's not what's happening right now). It should restart the uf2 bootloader.

Note that double clicking the reset button DOES drop into the bootloader.

I feel like we "fixed" this once before and it had to do with mismatch between the magic numbers or backup RAM locations used by the two parts (tinyuf2 and CP)

hathach, 2022-01-17:
It is probably fixed a while ago with https://github.com/adafruit/tinyuf2/pull/130 . But it requires to update 2nd stage bootloader. Could you try to re-flash the whole bootloader using the `combined.bin` to see if that works. 

note: drag & drog self-update uf2 only update the **"tinyuf2 app"**

jepler, 2022-01-17:
I'll check with the tinyuf 0.8.0 release and with https://learn.adafruit.com/adafruit-esp32-s2-tft-feather/factory-reset and report back -- I think/thought the board was in "as shipped by adafruit" condition so it would be nice to make sure that does work.

jepler, 2022-01-17:
A factory fresh Feather ESP32-S2 TFT INFO_UF2.TXT has
```
TinyUF2 Bootloader cd9d938 - tinyusb (0.10.1-109-g9b3ec69b)
Model: Adafruit Feather ESP32-S2 TFT
Board-ID: ESP32S2-FeatherTFT-revA
Date: Dec 17 2021
```
This does work properly:
```
>>> microcontroller.on_next_reset(microcontroller.RunMode.UF2)
>>> microcontroller.reset()
```
And this sends it all the way to the immutable ROM bootloader:
```
>>> microcontroller.on_next_reset(microcontroller.RunMode.BOOTLOADER)
>>> microcontroller.reset()
```

However, a script intended to _remotely_ trigger it just restarts CircuitPython (tickle.py):
```shell
#!/usr/bin/python
import sys
import serial
s = serial.serial_for_url(sys.argv[1])
s.baudrate = 1200
s.dtr = False
```
Run like:
```
$ tickle /dev/ttyACM1
```
just restarts CircuitPython. It gives the same blink pattern as a click of the reset button, the red LED and the Neopixel light up (the latter seems to be red+blue), then the 3 orange flashes from CP indicating the moment to activate safe mode.

So my testing had a couple of bits of confusion, and I think what is not working is the 'serial port' trick.

PR #5880 (2022-01-21): Request ROM bootloader when resetting using the 'serial trick'
Closes: #5870.
closed 2022-01-21 by microDev1