adafruit/circuitpython #5882: Read from displays (SPI or parallel)

kmatch98, 2022-01-18:
There are some situations where you may want to read data from a display:

- verify the code and hardware is working, and the display is connected properly
- reduce microcontroller memory usage by storing data on the display rather than in a large buffer (for example, interactions with the display could be “non-displayio”, possibly with display drivers that have hardware acceleration)
- Creative visual or display interaction projects

**Note:** With SPI displays, they will need to have the MISO (sometimes called SDO) pin connected to the microcontroller

### desired functions
- `readPixel(x,y)`: returns an RGB value of the pixel at location `x,y`
-  `readRect(x1,y1,x2,y2,buffer)`: places into the buffer the RGB values of the pixels in the rectangular boundary specified by `x1,y1` and `x2,y2`

## CircuitPython additions required

To enable reading from displays in CircuitPython will require additions to the core. Here is my first look at what might be required:

### displayio (also see questions below)

- add functions for `readPixel` and `readRect` - these are the high level commands that the user would normally use. 
- SPI “FourWire” and “paralleldisplay” - add read functions to support the high level displayio functions

### Additions to display drivers

Add the specific display driver chip’s command constants that are sent to the display to trigger the `readPixel` or `readRect` operation

### error handling
- For SPI, it’s desirable to provide an error if the MISO pin is not connected (if not connected, does a read operation return all zeroes or ones?)
- If the display driver isn’t updated with the chip-specific Read constant commands, it should error that “this driver chip library is not configured for Read operations”. 

## Questions
- A CircuitPython display doesn’t necessarily need to run using displayio.  Should the display `readPixel` and `readRect` commands fall into a separate module, unrelated to displayio?

ladyada, 2022-01-18:
have you found any SPI displays that let you read pixel data? so far we've only seen this ability on the parallel interfaces

kmatch98, 2022-01-18:
Yep I’ve got it working in Arduino by adding some found readPixel code to the Adafruit TFT library. I’ve demoed readPixel on your [2.8” capacitive touch breakout](https://www.adafruit.com/product/2090) connected via SPI (this unit uses a ILI9341 driver chip) to the Metro M4 express. 

(Note: I haven’t tested on ST7789 displays.)

Attached demo video is grabbing the pixels and redrawing pixels at an x,y offset.

I see that many of the Adafruit all-in-one display/micro controller boards don’t connect the MISO pin (Clue, PyGamer, PyBadge, MonsterMask, Hallowing, FunHouse).  Perhaps a MISO pad could be added in future products to allow for full loop display hardware testing, and to catch display timing issues like https://github.com/adafruit/circuitpython/issues/4775

https://user-images.githubusercontent.com/33587466/149996666-2bce9be1-245e-4ca6-b038-8da5fbb920e3.mov

tannewt, 2022-01-18:
I think we may want to expand the lower level `send` capability of FourWire and ParallelBus instead of providing something higher level. That will minimize the amount of code we need to add to the core but still give you the access you need.