adafruit/circuitpython #5898: Broadcom bare metal multi core support

thezim, 2022-01-20:
Does the Broadcom build support multi core if available?

dronecz, 2022-01-20:
[Here ](https://www.youtube.com/watch?v=_cjx9SA5vLs&list=PLjF7R1fz_OOXBHlu9msoXq2jQN4JpCk8A&index=5&t=3501s)is the answer 🙂

thezim, 2022-01-21:
The plain text answer here is no. I feel this will always be the case until the old micropython version it is based on is bumped to a more current version.

tannewt, 2022-01-21:
@thezim We're currently based on MP 1.17 so we're only one version behind. (1.18 was *just* released.) That isn't the issue.

The issue is that multicore is complicated and not all that useful for CP. Folks can always use Linux and CPython if they need multicore.

I'm not going to rule out adding it eventually. It's just a low priority.

tannewt, 2022-11-16:
#7218 is discussing a coprocessor API that could apply to the broadcom port.