adafruit/circuitpython #5904: RX and TX in use on QTPY ESP32S2

Neradoc, 2022-01-22:
Trying to use board.TX or board.RX, or board.UART raises a pin in use error. For example like this:
```py
import board
uart = board.UART()
```
```
ValueError: TX in use
```
I assume that comes from #5894

tannewt, 2022-01-24:
This should only be the case on DEBUG=1 builds. Did you try a DEBUG=0 build?

Neradoc, 2022-01-24:
That is with latest from S3. It doesn't look like there is any condition other than the pins definitions existing in `supervisor/shared/serial.c`. There's a few places in the file that test it. Am I missing something ?

https://github.com/adafruit/circuitpython/blob/501017066c615d6a5e501c7a808bc6349a6b3603/supervisor/shared/serial.c#L62-L74

Should the pins definition be conditioned by the DEBUG level, or change it in serial.c with something like that ?
```C
#define CIRCUITPY_DEBUG_UART CIRCUITPY_DEBUG && defined(DEBUG_UART_TX) && defined(DEBUG_UART_RX)
```

tannewt, 2022-01-25:
Ah, you are right. Yes, I expect it to be conditioned on whether it is a debug build. I'm not sure where the best place is to do that.

PR #6056 (2022-02-18): free RX and TX on QTPY-ESP32S2 in non debug builds
Fix #5904
closed 2022-02-18 by tannewt