adafruit/circuitpython #5906: 7.2.0-alpha.1 I2C issues when scanning the bus

UnexpectedMaker, 2022-01-22:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0-alpha.1-189-g9a353a4f5-dirty on 2022-01-22; TinyS2 with ESP32S2
But really on any ESP32 board. S2 and S3.
```

### Code/REPL

```python
import board
import busio
i2c = busio.I2C(board.SCL, board.SDA)
while not i2c.try_lock():
    pass
    
[hex(x) for x in i2c.scan()]
```

### Behavior

Scan hangs for 60-90 seconds, and then comes back with nothing found.
Confirmed on my TinyS2, TinyS3, FeatherS3 and corroborated with @prplz using the esp32s3 devkitc n8r8

UnexpectedMaker, 2022-01-22:
Ok, if the slave has pull-ups.... it seems to work, if multiple slaves connected and one has pull-ups, it works and finds them all, but if the one with pull-ups is removed, the next scan long hangs again.

If a slave with no pull-ups is connected and nothing else,  it long hangs.

Only once I've see the "no pull-ups error".... every other time it hangs. I've plugged in and unplugged 4 different I2C modules, some with and some without pull-ups, I can't make sense out of what is happening.

prplz, 2022-01-22:
I did a little printf debugging and found that this for loop is taking about one second per iteration https://github.com/adafruit/circuitpython/blob/main/shared-bindings/busio/I2C.c#L143-L148

closed 2022-02-01 by tannewt

dhalbert, 2022-02-01:
@prplz @UnexpectedMaker This was closed by #5958, and I reduced a timeout, but reopen or make a new issue if you are still seeing slow scan times, and we need something like #5908 (or comment in that PR).

PR #5958 (2022-02-01): I2c common hal write read
I believe this fixes all these ESP32-S2 I2C issues:
- Fixes #5934.
- Fixes #5906.
- Fixes #5871.
- Fixes #5680.

As suggested by @tannewt in https://github.com/adafruit/circuitpython/issues/5680#issuecomment-1024511731, this PR provides a new `common_hal_busio_i2c_write_read()` operation. `shared-bindings/busio/I2C.c` no longer calls `write` and `read` separately itself, but calls the combined routine.
- On ESP32-S2, the `common-hal` I2C routines use the new ESP-IDF `i2c_master_device...()` wrapper routines, which set things up properly for I2C transactions.
- On othe [... cut]