adafruit/circuitpython #5915: Increase pystack and / or max sockets on ESP32-S3

anecdata, 2022-01-24:
The ESP32-S3 has 192KB more ram (512KB total) compared to the ESP32-S2 (320KB total). We may want to consider increasing pystack and possibly also the max number of sockets.

I think there is one case where this is done for a specific board, and I think it's possible to do for a port in the pot config files, but this would be for the ESP32-S3 chip only.

tannewt, 2022-01-24:
@anecdata Would you be willing to do this? I think you are better positioned to determine the best new values.

anecdata, 2022-01-24:
I can take a stab at it. It may take a while, my build environment has been down since the esp-idf update and I need to dedicate some time to rebuilding it from scratch. (if anyone gets impatient, have at it)

I don't have an intuition for the best new values, but I can test some incremental variations. 

I assume this feature use the `#if defined(CONFIG_IDF_TARGET_ESP32S3)` construct for chip-level as opposed to board- or port-level.

tannewt, 2022-01-25:
Ya, CONFIG_IDF_TARGET is how I'd do it in header files.

Yes, the main thing I want to leave some RAM for is BLE. I'm not sure how much it'll use.

anecdata, 2022-02-09:
Also may be able to increase the maxmimum number of connections a wifi AP can accept (currently 4).

anecdata, 2022-02-11:
esp-idf 4.4 improved internal SRAM memory allocation from what I could see. Pre-4.4, applications would run down the largest free block by powers of 2, and due to fragmentation it could quickly get down to into 3-figures (e.g., 256 bytes), and would sometimes run out of memory. Post-4.4, allocations seem to be more finely-grained, and largest free block stays in healthy 5-figure territory even for demanding applications on the ESP32-S2.

Here's an example of idf memory used by a `wifi` `Monitor` application after running for some time. Applications like this are heavy users of idf memory due to allocation of a 128-(or more)-element queue for wifi frames. In this snapshot, the queue is about half full in each case, meaning ~64 frames are fully in memory (frames can theoretically go over 2K, but they are typically <300 bytes):
```
         espidf.heap_caps_get_*:

         largest_free_block()  free_size()  total_size()
ESP32-S2                14848        38092        158404
ESP32-S3               196608       203564        344732
```
As expected, the 192KB increase in onboard memory on the S3 is reflected in the key measures we have of running idf memory state.

I think it's relatively safe to conservatively increase some of the parameters set in the core, specifically for the S3, for features that use idf memory, while also keeping a healthy chunk in reserve for BLE or other features:

- pystack (currently 1536 bytes)
- maximum number of network sockets (currently 4, resulting in max of 2 simultaneous HTTPS connections)
- maximum number of stations that can connect to a `wifi` Access Point (currently 4)

Please chime in if I'm missing anything. I'll proceed to test some variations and see how they reflect in our `espidf.heap_caps_get_*` measures.

anecdata, 2022-02-11:
1. Looks like `CIRCUITPY_PYSTACK_SIZE` default is set in `py/circuitpy_mpconfig.h`, and could be changed in `ports/espressif/mpconfigport.h`, conditioned on `CONFIG_IDF_TARGET_ESP32S3`.
2. `CONFIG_LWIP_MAX_SOCKETS=4` is set in `ports/espressif/esp-idf_config/sdkconfig.defaults`. I'm not sure how to cleanly make that conditional on ESP32S3. Maybe override it in `ports/espressif/esp-idf-config/sdkconfig-esp32s3.defaults`?
3. Max connections to an Access Point I think is easy. I'd propose making it a kwarg in the AP init. I left a comment in the AP code to that effect. Users could end up with memory errors if they overdo it.

Another issue: I don't think we want to change anything for boards with no external PSRAM, since they'll be tight using internal for everything. `ports/espressif/supervisor/port.c` already bumps `HEAP_SIZE` from 48KB to 176KB on ESP32S3 "for when there is no external RAM for CircuitPython to use exclusively". But SPIRAM is a board-specific feature. Is there a better way than to make any changes also dependent on `CONFIG_ESP32S3_SPIRAM_SUPPORT=y` as set in `ports/espressif/boards/*/sdkconfig`? I'm not entirely clear on the order all these files get checked., but it seems boards override ports, and ports override circuitpython global.

tannewt, 2022-02-11:
> I'm a little concerned about making spaghetti conditionals criss-crossed over so many files.

Ya, agreed. I haven't split out an S3 with RAM bucket yet. Maybe just increase max sockets on all S3? Hopefully we still leave enough room for it after the increased heap size for boards without RAM.

The sdkonfig ordering comes from here: https://github.com/adafruit/circuitpython/blob/main/ports/espressif/Makefile#L317

jepler, 2022-07-06:
Feel free to open a separate issue about tweaking the pystack.

closed 2022-07-06 by jepler