adafruit/circuitpython #5916: Bitmap.blit from OnDiskBitmap => hard crash

rsbohn, 2022-01-24:
### CircuitPython version

```python
Adafruit CircuitPython 7.1.0 on 2021-12-28; Adafruit FunHouse with ESP32S2
```

### Code/REPL

```python
import displayio
with open("/bmp/allblue.bmp","rb") as bits:
    source = displayio.OnDiskBitmap(bits)
    dest = displayio.Bitmap(source.width, source.height, 64)

dest.blit(0,0, source)
```

### Behavior

Hard reset of FunHouse board.

### Additional information

May not be possible, but the response should be an error message rather than a hard crash.

rsbohn, 2022-01-24:
[allblue.zip](https://github.com/adafruit/circuitpython/files/7928123/allblue.zip)
Here is the bitmap used in the script.

FoamyGuy, 2022-01-24:
I think bitmaptools.blit() is expecting to operate on a "normal" `Bitmap` object rather than an `OnDiskBitmap`.

If you are using it with a bmp file I think the file needs to be loaded with `adafruit_imageload`

We should definitely have a better error mitigation and avoid the hard crash though.

jepler, 2022-01-24:
It looks like `Bitmap.blit` intends to only accept another `Bitmap` but has insufficient error checking. We should fix this bug so that a TypeError exception is raised.

Also, while it may have been suggested by a linter, this is not correct usage:
```python
with open("/bmp/allblue.bmp","rb") as bits:
    source = displayio.OnDiskBitmap(bits)
```
because the OnDiskBitmap object needs to retain an open file for its whole lifetime, this won't work properly. Instead, the `source` bitmap becomes unusable at the end of the `with` statement. You'd instead just want to write:
```python
source = displayio.OnDiskBitmap(open(...))
```
or separate it into two lines, as the documentation suggests:
```python
odb = displayio.OnDiskBitmap('/sample.bmp')
face = displayio.TileGrid(odb, pixel_shader=odb.pixel_shader)
```

rsbohn, 2022-01-24:
I should have put the `.blit()` inside the `with` form. For normal usage you are correct, but if I can copy the bits to a second bitmap the original file is no longer needed.

closed 2022-02-15 by dhalbert

rsbohn, 2022-02-15:
Thanks for working on this issue.

PR #6035 (2022-02-15): bitmap.blit type checking and raise error
resolves #5916 I confirmed the hard crash as described in the issue.

Tested on Feather TFT ESP32-S2 using this example code (mostly the same as posted in the issue):
```py
import time
import displayio

print("waiting 8...7...6..5...4...3...2...1...")
time.sleep(8)

with open("/allblue.bmp", "rb") as bits:
    source = displayio.OnDiskBitmap(bits)
    dest = displayio.Bitmap(source.width, source.height, 64)
    dest.blit(0, 0, source)

```

output after this change:
```
code.py output:
waiting 8...7...6..5...4...3...2...1...
Traceback (most recent call [... cut]