adafruit/circuitpython #5924: High Deep Sleep current with QT Py ESP32-S2

alanmitchell, 2022-01-25:
### CircuitPython version

```python
Primarily tested on CP 7.1.1, but also tested on CP 7.1.0, CP 20220123-4d0e1d9, and CP 6.3 using the Saola WROVER UF2 file.
```

### Code/REPL

```python
import alarm
import time
import board
from digitalio import DigitalInOut, Direction

# turn off power to the Neopixel
#neo_pwr = DigitalInOut(board.IO38)   # use with CP 6.3 Saola UF2
neo_pwr = DigitalInOut(board.NEOPIXEL_POWER)
neo_pwr.direction = Direction.OUTPUT
neo_pwr.value = False

time_alarm = alarm.time.TimeAlarm(monotonic_time=time.monotonic() + 5)
alarm.exit_and_deep_sleep_until_alarms(time_alarm)
```

### Behavior

The code executes correctly, but the Deep Sleep current, as measured by a Nordic Power Profiler Kit II (PPK II), is 5 mA.  I was expecting Deep Sleep current of 200 uA or less.  I am powering the QT Py from the PPK II at about 4.0 Volts and applying that voltage to the Bat and Gnd pads.  There is no USB cable connected to the board.

### Additional information

I have tested this with two different QT Py ESP32-S2 boards with the same result.  I have also run the same script on an Unexpected Maker Feather S2 board (running CP 6.3) and measured a Deep Sleep current of 70 uA, which helps validate my script and measurement setup.

tannewt, 2022-01-25:
Please try a 7.2.0 build via `Absolute Newest` on the site as well. It's possible we've tweaked the board's behavior in main already.

alanmitchell, 2022-01-25:
I just re-checked the S3 site for the Absolute Newest, and it appears to be the one I already tested, as mentioned in the Circuit Python version section above (CP 20220123-4d0e1d9):

[image]

alanmitchell, 2022-01-25:
Here is the screenshot from the Power Profiler Kit II, the deep sleep period visible at the right side of the trace:
[image]

tannewt, 2022-01-26:
Ah, sorry I missed that. Thanks for trying it.

dhalbert, 2022-01-26:
I reproduced this and am working on it.

alanmitchell, 2022-01-26:
Thanks Dan, glad it wasn't a goof on my part, but hope you can fix it...

Alan Mitchell
Analysis North <https://analysisnorth.com>
Cell: 907-310-9124

dhalbert, 2022-01-27:
It turns out this is a hardware problem specific to the QT Py ESP32-S2. We know how to fix it, but it's not an easy trace-cutting or something like that. For now, you may want to switch to a different board. We could get somewhat less sleep current by using ESP-IDF "light sleep" (which is not the same as CircuitPython's light sleep), but it would still be several mA.

alanmitchell, 2022-01-27:
OK, thanks for running this down.  For the current application, I can live with 5 mA, but uA sleep would be good in the future.  But, the long boot up time for CircuitPython limits the number of low-power applications where it can be used.

alanmitchell, 2022-01-27:
FWIW, can you share a quick explanation of the hardware issue (I'm an EE by degree)?

dhalbert, 2022-01-27:
The VDD_SPI pin was unintentionally pulled high, but should have been left unconnected except for the filter caps on it. Compare the schematics of the QT Py and the Feather ESP32-S2. The Feather uses a module in a can; the QT Py has bare parts. The module has the same chip inside (an ESP32-S2 package that includes PSRAM and flash in a single package.)

Re long boot-up time: we wait five seconds to see if USB has been connected, and allow for enumeration. But we could conceivably _not_ wait for USB: that could be a sleep option. The problem is then how to get back to the REPL or CIRCUITPY if you are in a sleep cycle: you'd need to get into safe mode, or force-erase the flash.

alanmitchell, 2022-01-27:
Thanks for the explanation of both issues. Taking 5 seconds off the boot up time would be huge for low-power stuff; hope you can eventually come up with a solution for getting back to CIRCUITPY after that.  I'll Close this issue.

closed 2022-01-27 by alanmitchell

tannewt, 2022-01-27:
The boot up time should only be 5 seconds on power up or reset press, not wake from deep sleep. Can we not tell the difference on the S2? I'm pretty sure it used to work on the MagTag.

dhalbert, 2022-01-27:
I put in the five second wait because otherwise it can be very hard to ctrl-C a program that wakes up, checks something, and immediately goes back to sleep. When I didn't have this wait, my deep-sleep test programs were almost impossible to interrupt.

tannewt, 2022-01-27:
So the wait is in `code.py`? I don't see one in the first post here.

My assumption was that folks would hit reset to get back it to slow down. When waking up from deep sleep it should be fast. For debugging, folks should connect usb and then use fake deep sleep.

dhalbert, 2022-01-27:
The wait is here:
https://github.com/adafruit/circuitpython/blob/3904421a22a8185a5904a96f433374241ac38f5d/main.c#L530-L540

The problem is that when we start code.py, we have no idea whether we are connected to USB or not, because the enumeration can take a few seconds. So we can't decide whether to do a real or a fake deep sleep. We might get to the `exit_and_deep_sleep()` call in code.py very quickly, before enumeration has happened. So we don't make the real/fake deep sleep decision until enough time has gone by for enumeration to complete.

tannewt, 2022-01-28:
Ah, we should only do that if we woke due to reset or power on. We should skip it if waking from deep sleep. If we woke for deep sleep, then we should assume we want to continue deep sleeping. That'll save the most power.

Want me to reopen this issue or make a new one?

dhalbert, 2022-01-28:
Yes, let's make a new issue. So on the first sleep, we don't know, so that will be at least 5 seconds. After that, if the user connects USB _and presses reset_, then it will go into fake deep sleep, and be interruptible. That is a bit of change, since right now just connecting USB is enough (I think).

alanmitchell, 2022-01-28:
Thank you all for pursuing this; it will really improve low-power performance.  5 seconds at 55 mA every time you wake up to read a sensor is a serious hit on the battery.

muendlein, 2022-02-03:
> It turns out this is a hardware problem specific to the QT Py ESP32-S2. We know how to fix it, but it's not an easy trace-cutting or something like that. For now, you may want to switch to a [...]

I seriously have to question the QA procedures because something like should have never happened in the first place.

ladyada, 2022-02-03:
@muendlein if ya cant use the board, we can refund it. future boards will have the fix and we'll get screenshots of low power usage to verify for new designs!

ladyada, 2022-02-28:
Rev C has been fixed and will be available shortly. i just verified the draw in deep sleep with neopixel turned off is 80uA

[image]

alanmitchell, 2022-02-28:
Thanks, much appreciated!  And isn't that PPK II great...