adafruit/circuitpython #5930: M4 Express enters safe mode

rp10007, 2022-01-27:
### CircuitPython version

```python
Adafruit CircuitPython 7.1.0 on 2021-12-28; Adafruit ItsyBitsy M4 Express with samd51g19
Board ID:itsybitsy_m4_express
```

### Code/REPL

```python
The code that apparently makes things fall down is in a file called twoWheelHelper.py:

#Two-wheel with driver

import board
import time
import pwmio
from adafruit_motor import servo, motor
import digitalio
import countio
import pulseio

bracket = (servo.Servo(pwmio.PWMOut(board.D2, duty_cycle=2**15, frequency=50)))
#bracket.angle = 90
import adafruit_hcsr04 as sonar
trig = board.D9
echo = board.D7
ears = sonar.HCSR04(trig, echo)
#pin_counter = countio.Counter(board.D1, edge=Edge.RISE, pull=digitalio.Pull.UP)

#counter1 = countio.Counter(board.MISO)
#counter2 = countio.Counter(board.MOSI)

#encoder1 = digitalio.DigitalInOut(board.MISO)
#encoder1.pull =  digitalio.Pull.UP

pulses1 = pulseio.PulseIn(board.MISO, maxlen=100)
pulses2 = pulseio.PulseIn(board.MOSI, maxlen=100)

#encoder2 = digitalio.DigitalInOut(board.MOSI)
#encoder2.pull =  digitalio.Pull.UP
#encoders currently reversed, fix
m1a = pwmio.PWMOut(board.D10, frequency=50)
m1b = pwmio.PWMOut(board.D11, frequency=50)
motor1 = motor.DCMotor(m1b, m1a)
m2a = pwmio.PWMOut(board.D12, frequency=50)
m2b = pwmio.PWMOut(board.D13, frequency=50)
motor2 = motor.DCMotor(m2b, m2a)

def dist(sonar):
    try:
       foo = sonar.distance
       return foo
    except:
        return 9999

def waitsonar(d=10):
    while dist(ears) < d or dist(ears) == 9999:
        time.sleep(1)

def forward(duration, speed):
    motor1.throttle = speed
    motor2.throttle = speed
    time.sleep(duration)
    motor1.throttle = None
    motor2.throttle = None

def fwdsonar(duration, speed, distance):
    now = time.monotonic() # start the clock
    motor1.throttle = speed
    motor2.throttle = speed
    while time.monotonic() < now + duration:
        if dist(ears) < distance:
            motor1.throttle = None
            motor2.throttle = None
            return "prox"
        time.sleep(0.1)
    motor1.throttle = None
    motor2.throttle = None
    return "time"

def fwdprinting(duration, speed):
    total1 = 0
    total2 = 0
    oldval1 = encoder1.value
    oldval2 = encoder2.value
    now = time.monotonic() # start the clock
    motor1.throttle = speed
    motor2.throttle = speed
    while time.monotonic() < now + duration:
        if encoder1.value != oldval1:
            total1 += 1
            print("1=", total1)
            oldval1 = encoder1.value
        if encoder2.value != oldval2:
            total2 += 1
            print("2=", total2)
            oldval2 = encoder2.value
        if total1 + total2 > 10:
            motor1.throttle = speed*total2/total1
            motor2.throttle = speed*total1/total2
            print(motor1.throttle, motor2.throttle)
    motor1.throttle = None
    motor2.throttle = None
    return "time"

def backward(duration, speed):
    motor1.throttle = -speed
    motor2.throttle = -speed
    time.sleep(duration)
    motor1.throttle = None
    motor2.throttle = None

def spin (direction, duration, speed):
    motor1.throttle = None
    motor2.throttle = None
    if direction in ['left', 'counterclockwise', 'widdershins']:
        motor1.throttle = -speed
        motor2.throttle = speed
    elif direction in ['right', 'clockwise']:
        motor1.throttle = speed
        motor2.throttle = -speed
    time.sleep(duration)
    motor1.throttle = None
    motor2.throttle = None

def maneuver(duration, speed1, speed2):
    motor1.throttle = speed1
    motor2.throttle = speed2
    time.sleep(duration)
    motor1.throttle = None
    motor2.throttle = None

def listfuncs():
    print("""def dist(sonar): returns sonar val catching error as 9999
def forward(duration, speed):
def backward(duration, speed):
def spin (direction, duration, speed): 'left' or 'right'
def maneuver(duration, speed1, speed2): control motors individually""")

imported by code.py

# code for ItsyBitsyM4 Express on two-wheel, two-motor body with servo-mounted sonar and yourduino motor driver

from twoWheelHelper import *

spd = 2.0/360.0 # seconds per degree, decimal point so python won't round to int

def collectsonar(increment=10):
    distances = {}
    bracket.angle=0
    time.sleep(0.3)
    for degrees in range(0, 181, increment):
        bracket.angle = degrees
        time.sleep(0.1)
        distance = dist(ears)
        distances.update({degrees : distance})
    bracket.angle = 90
    return distances

def pickangle(sonardict):
    maxDistAngle = max(sonardict, key = sonardict.get)
    maxDist = sonardict[maxDistAngle]
    return ([maxDistAngle, maxDist])

# the following relies on some measurement and geometry
# wheels are about 65mm, so one revolution of the wheel is about 20 cm
# quick measurement says about 0.4 second for one revolution
# wheels are about 70 mm out from center, so a little more than two turns for 360.
# we can start with 1 second for 360 and make measurements.

def turnToAngle(angle):
    if angle < 90:
        angle = 90 - angle
        spin('right', angle*spd,1)
    elif angle > 90:
        angle = angle - 90
        spin('left', angle*spd, 1)

def simpleavoid(duration, distance, direction, angle):
    now = time.monotonic()
    while time.monotonic() < now + duration:
        forward(0.1, 1)
        print(dist(ears), distance)
        if dist(ears) < distance:
            print('should spin', angle*spd, 'seconds')
            spin(direction, angle*spd,1)

def sonaravoid(duration, distance, direction, angle):
    now = time.monotonic()
    while True:
        result = fwdsonar(duration, 1, distance)
        if result == "time":
            return
        spin(direction, angle*spd, 1)

bracket.angle=90
#waitsonar(20) # we do this so we can set the bot down and not have it start immediately
print('finished waiting')

#simpleavoid(20, 25, "right", 90)
```

### Behavior

When I  do "from twoWheelHelper import *" at the REPL, I lose the USB connection, and  the serial connection goes away. When the connection comes back (sometimes but not always) I get the following:

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
Please file an issue with the contents of your CIRCUITPY drive at 
https://github.com/adafruit/circuitpython/issues

At that point a keystroke usually re-enters the REPL, which usually appears to have basic function. Sometimes Mu/REPL just freezes. 

### Additional information

I had thought this might be a result of marginal power, but it happens whether the motor driver is on or off (switched). Tried with a direct USB connection and a powered USB port, no apparent difference.

 I've also been playing around a lot with pulseio and countio, and it appears that only certain pins yield working Pulsein or counter, but on the M4 I haven't been able to figure out which ones. (the 2040 port gives error messages, on the M4 certain choices just silently do or don't work). No idea whether that's involved in this or it's just some weird hardware thing.

rp10007, 2022-02-03:
I don't know if this is part of the crash behavior, but on the countio side I have discovered an odd thing about the M4 port that isn't true elsewhere afaik: if I define a counter in a file that I then import (e.g. encoder = countio.Counter(board.MISO)), that counter doesn't work. It uniformly reports 0. If I deinit that counter in the repl and define a new counter on the same pin, it works fine. If I define the counter in code.py directly instead of defining it in in an imported file, it works fine. Same line of code, and I am doing "from helperfile import *" to do the import. Or am I just missing something?

dhalbert, 2022-02-03:
@rp10007 Could you open a new issue about the `countio` problem? Thanks!

dhalbert, 2022-02-09:
The `countio` bug in #5982 could be incrementing random memory, which could cause this crash.

dhalbert, 2022-03-22:
See if `PulseIn` needs to be long-lived also.

jepler, 2022-09-21:
We believed this was fixed due to #6384.Feel free to re-open or file a fresh issue if you still encounter the problem.

closed 2022-09-21 by jepler