adafruit/circuitpython #5931: Espressif: GPIOs default state is HIGH

MicroDev1, 2022-01-27:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0-alpha.1-229-gfff68c9f8 on 2022-01-27; MicroDev microS2 with ESP32-S2
```

### Code/REPL

```python
N/A
```

### Behavior

I noticed that the on-board led was dimly lit and on further investigation, I found that the GPIOs were at ~3.2V.

tannewt, 2022-01-27:
Yes, I did this deliberately in #5892 to keep the pins from floating (specifically neopixel pins.) Perhaps we need to special case on-board LED pins.

dhalbert, 2022-02-08:
As noted in #5679:
> If the GPIO is placed in InputEnable mode (without pull up or down, i.e., just create a DigitalinOut object for the pin), it happens to stop the flickering. But since the neopixel pin is usually freed by the supervisor between color settings, it is normally placed in input disabled / output disabled / no pulls mode.

So maybe the solution is to put it input mode, instead of defaulting to high? I think this is more like SAMD, etc., where the default state is input.

tannewt, 2022-02-08:
Setting pins to input mode is a higher power consumption than a pull. See https://www.st.com/resource/en/application_note/dm00315319-stm32-gpio-configuration-for-hardware-settings-and-low-power-consumption-stmicroelectronics.pdf chapter 6. Floating pins with input enabled will randomly flicker the input buffer on and off and consume power. It's better to pull each pin that is unused.

However, for cases like this, I think we want to have boards dictate whether pins are pulled high or low.

dhalbert, 2022-02-17:
Another side effect of the default high is that, at least on the MagTag, it cycles the speaker amp, which causes clicking and a quiet hum.

dhalbert, 2022-02-17:
In some discord discussion, @tannewt suggested that we needed to be able to special-case certain pins, for reasons such as the above.

PR #6057 (2022-02-18): Allow ESP boards to customize how a pin is reset
This allows board code to override the default pull up reset state.

It is useful for pins that are already externally connected, pulled
or otherwise used by the board.

Fixes #5931
commit c4fb5f7d (2022-02-18): Allow ESP boards to customize how a pin is reset

This allows board code to override the default pull up reset state.

It is useful for pins that are already externally connected, pulled
or otherwise used by the board.

Fixes #5931
closed 2022-02-18 by dhalbert