adafruit/circuitpython #5932: white noise with neotrellis and 7.x version

durandom, 2022-01-27:
There seems to be a regression from 5.x to 7.x

Theres a white noise / static coming, when no sound is played.

see the last post in https://forums.adafruit.com/viewtopic.php?f=57&t=177457&p=864847 
downgrading to 5.x solves the issue for me as well. 
I recreated it with the drum machine demo, which has a 5 and 7 compatible version

not sure if this more an issue for the circuit python community or this one.

durandom, 2022-01-27:
Thanks @dhalbert 
I came up with a small reproducer in https://github.com/durandom/neotrellis

* the code in noise/7 produces a static white noise with a Neotrellis M4 whereas the code in noise/5 does not.
* 5 is using `adafruit-circuitpython-trellis_m4_express-en_US-5.1.0.uf2`
    * `adafruit-circuitpython-bundle-5.x-mpy-20201205`
* 7 is using `adafruit-circuitpython-trellis_m4_express-en_US-7.1.1.uf2`
    * `adafruit-circuitpython-bundle-7.x-mpy-20220127`

```python
import time
import board
import audioio
import audiomixer
import adafruit_trellism4

# Our keypad + neopixel driver
trellis = adafruit_trellism4.TrellisM4Express(rotation=90)
audio = audioio.AudioOut(board.A1, quiescent_value=50)
mixer = audiomixer.Mixer()
audio.play(mixer)

while True:
    time.sleep(0.5)
    print("tick..")
    audio.pause()
```

durandom, 2022-01-27:
IMHO this is a serious issue, since it really spoils the fun. It's a loud white noise static. And it's definitely related to the software stack

dhalbert, 2022-11-23:
I was unable to reproduce the problem with your test program on 7.3.3 or 8.0.0 beta(s). I have a powered speaker (Bluetooth speaker with aux input) connected to a Trellis M4. However, I tried a different test program that just plays mono wav files in sequence. I note that if I use stereo output and play mono files, every other play has a buzzing noise superimposed.  The test program is a simplification of https://learn.adafruit.com/abc-soundboards-with-neotrellis/code: the important part is:

```py
#audio = audioio.AudioOut(board.A0)
#audio = audioio.AudioOut(board.A1)
audio = audioio.AudioOut(board.A1, right_channel=board.A0)

while True:
    for sample_num in range(len(SAMPLES)):
        try:
            filename = SAMPLE_FOLDER+SAMPLES[sample_num]
            print(filename)

            with open(filename, "rb") as f:
                wav = audiocore.WaveFile(f)

                audio.play(wav)
                while audio.playing:
                    time.sleep(0.1)
        except OSError:
            pass # File not found! skip to next
```

I will try to fix the buzzing, but I think that's not the problem you had.

dhalbert, 2022-12-07:
@durandom I cannot reproduce your issue, as I mentioned. If you are still tracking down this problem, could you try the latest version of CircuitPython? There will be an 8.0.0-beta.5 shortly and there are relevant changes in the "Absolute Newest" version on circuitpython.org already. I will close this issue for now but we can reopen it or open a new one.

closed 2022-12-07 by dhalbert