adafruit/circuitpython #5933: Release -stub packages for CircuitPython-only libraries

tannewt, 2022-01-27:
A number of board-specific libraries aren't released to pypi. This makes sense for use because they require specific hardware. However, it makes it harder to get IDE auto-complete. To make IDE's happier, we could release -stub versions that just provide type info, not implementation.

Michael on Discord did it manually:

> i have a workaround, i downloaded https://github.com/adafruit/Adafruit_CircuitPython_MacroPad/blob/main/adafruit_macropad.py, pip installed mypy, then did stubgen adafruit_macropad.py then move the resulting out/adafruit_macropad.pyi to the root of my circuitpy drive

Based on finding `setup.py.disabled` these are the libraries that could be done:

- [ ] circuitplayground
- [ ] clue
- [ ] monsterm4sk
- [ ] nunchuk
- [ ] ov2640
- [ ] ov5640
- [ ] ov7670
- [ ] pyportal
- [ ] boardtest
- [ ] dash_display
- [ ] funhouse
- [ ] macropad
- [ ] magtag
- [ ] matrixportal
- [ ] pybadger
- [ ] pyoa

CharString, 2022-01-29:
The de facto naming convention for these typing stub packages on pypi is `f"types-{name_of_full_packages}"`

tannewt, 2022-01-31:
Thanks @CharString! Happy to match the convention.

tbjers, 2022-03-10:
Obligatory bump +1.

tekktrik, 2022-08-25:
@adafruit/circuitpythonlibrarians I think this should be closed but want final confirmation.  The board specific libraries are now available on PyPI for Mu, so typing should be available as they get type annotated.  Each of the repos has their own type annotation issues, so this is duplicate now, I think.

dhalbert, 2024-03-08:
Closing as requested.