adafruit/circuitpython #5938: Unable to Parse Valid Cpython syntax when concatenating f-strings

ewwaller, 2022-01-28:
### CircuitPython version

```python
Adafruit CircuitPython 7.1.1 on 2022-01-14; Adafruit Feather M4 Express with samd51j19
```

### Code/REPL

```python
Adafruit CircuitPython 7.1.1 on 2022-01-14; Adafruit Feather M4 Express with samd51j19
>>> x = (f"Hello "
... f"world")
Traceback (most recent call last):
  File "<stdin>", line 2
SyntaxError: invalid syntax
>>> x = (f"Hello " +
... f"world")
>>> x
'Hello world'
>>> x = (f"Hello"
... "world")
>>> x
'Helloworld'
>>> x = ("hello "
... f" world")
>>> x
'hello  world'
>>>
```

### Behavior

The parser does not properly concatenate two f-strings as used when splitting across lines.
One regular string and one f-string works correctly,
One g-string and one regular string works correctly
Two regular strings work correctly.
Two f-stings, "invalid syntax"
Using the "+" operator on two f-stings works, but it allocates a transient sting on the fly which then needs to be garbage collected 

Cpython accepts two f-srings 

### Description

Gets flagged at compile time as an error and code does not run.
REPL has same issue.

### Additional information

Here is the REPL session in cpython on Arch Linux working as expected
```
waller@odin/home/ewaller % python
Python 3.10.1 (main, Dec 18 2021, 23:53:45) [GCC 11.1.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> x = (f"Hello "
... f"world")
>>> x
'Hello world'
>>> 
```

jepler, 2022-01-28:
It's worth noting that this limitation also exists in micropython, see
 * https://github.com/micropython/micropython/issues/7858

```python
MicroPython v1.18-56-g517e82eb6-dirty on 2022-01-28; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> f"" f""
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
```
Using "+" instead of f-string literal concatenation is the official recommendation. If a future version of Micropython improves in this area, we'll get the improvement by merging that future release into CircuitPython. We should make sure that this language difference is mentioned in our docs, as it is in theirs.

ewwaller, 2022-01-28:
Thanks,  I had not searched the micropython side.
As I mentioned, the + operator creates an temporary immutable string for the first one, and then creates a second immutable string to store the result, and then frees the temporary string.  Just trying not to exercise gc -- especially on an embedded system.

I have been using this feather with Rust (RTIC) and Tinygo as circuitpython was not quite fo there when I first tried it.  In the mean time,  With async/await now on the scene, I am all in and look forward to helping where I can.

tannewt, 2022-01-28:
> I have been using this feather with Rust (RTIC) and Tinygo as circuitpython was not quite there when I first tried it. In the mean time, With async/await now on the scene, I am all in and look forward to helping where I can.

Awesome! You may want to join #circuitpython-dev on the Discord chat as well: https://adafru.it/discord That's where we discuss CP dev work.

jepler, 2022-01-29:
It looks like the very useful "core language" page from micropython is not in the circuitpython docs: https://docs.micropython.org/en/latest/genrst/core_language.html

most of these items apply to circuitpython as well.