adafruit/circuitpython #5941: Adafruit Feather ESP32-S2 TFT - TFT_BACKLIGHT in use

anecdata, 2022-01-29:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0-alpha.1-224-gac7a80753 on 2022-01-26; Adafruit Feather ESP32-S2 TFT with ESP32S2
```

### Code/REPL

```python
>>> import board
>>> import pwmio
>>> 
>>> pwmio.PWMOut(board.TFT_BACKLIGHT, frequency=5000, duty_cycle=0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: TFT_BACKLIGHT in use
```

### Behavior

Similar to #5903. I'd expect that independent control of the backlight should be possible in user code.

tannewt, 2022-01-31:
Why is `board.DISPLAY.brightness` insufficient?

Sharing a pin between the display and pwmio is tricky internally and I'm trying to avoid us having to do it. (The status neopixel rework made this much simpler.)

anecdata, 2022-01-31:
I can live with the brightness control.

It was the change from prior, and another "pin in use" scenario that wasn't intuitive (to me). The use cases are probably limited (for example, futzing with PWM to make a glitchy-looking display).

If there's isn't a compelling reason to override the complexity in the core, we can close this.

tannewt, 2022-02-01:
What version did it work in? Sometimes in early ports we don't have proper pin in use checking.

anecdata, 2022-02-01:
I don't recall what I had on originally. Closing this. Someone can re-open or make new if there's a compelling use case.