adafruit/circuitpython #5956: Safe Mode: mechanism for user code to recover without manual intervention

anecdata, 2022-01-31:
Feature request based on some discussion in this issue:
https://github.com/adafruit/circuitpython/issues/2694#issuecomment-740624686

It would be very useful to have a mechanism for user code to intercept or detect Safe Mode, and trigger its own reload or reset. Or, have a user-settable indicator not to go into Safe Mode for at least some causes. Ideally we'd want to identify and fix every last intermittent software cause of Safe Mode, but sometimes they are rare  and very difficult to debug, and perhaps triggered from code included from some port-specific development environment.

For now, I have a microcontroller watchdog that looks for a heartbeat signal from the main microcontroller and resets it if it's lost. A software solution would simplify deployments, particularly for devices that are placed in less-accessible locations.

tannewt, 2022-02-01:
Would a `safemode.py` work in your case? How does this interact with the internal watchdog?

Safe mode is really meant as a "we don't know what went wrong and don't expect to recover". Anything recoverable should be done through exceptions instead.

anecdata, 2022-02-01:
`safemode.py` would work fine. Anything that the user can control.

`watchdog` module doesn't catch these since the code stops feeding it, nor do Exceptions. I'm not sure if there would be any interactions / complications from having both. Nothing comes to mind.

Most (all?) of my Safe Mode occurrences are intermittent but fixed with a restart. 95% sure it's not power or hardware. The idea is that the device can self-recover with a software restart.

mrdalgaard, 2022-02-02:
Maybe a safemode.py could also be used to account for brownout safemode occurrences in battery/solar powered projects.

maholli, 2022-02-14:
+1

Still eagerly awaiting safemode.py! :)

adamwolf, 2022-03-11:
Where would we want to put the `safemode.py`?  Next to `code.py` and `boot.py` et al?

I looked into implementing this, or at least starting to, and I quickly ran into a comment in main.py 

```
     // Create a new filesystem only if we're not in a safe mode.
     // A power brownout here could make it appear as if there's                                                                               
     // no SPI flash filesystem, and we might erase the existing one.
```

tannewt, 2022-03-14:
@adamwolf Yes, I'd do it like code.py I think. That way serial output can go out usb potentially.

protosam, 2022-07-26:
What if the user locks themself out of the file system by disabling it and the serial? Perhaps the safemode should be an alternate binary?

dhalbert, 2022-07-26:
@protosam in the worst case, the user can load a .uf2 that erases the CIRCUITPY flash, e.g.: https://learn.adafruit.com/welcome-to-circuitpython/troubleshooting#erase-circuitpy-without-access-to-the-repl-3105309

I'm not sure what you mean by an "alternate binary", but in general there's not enough space for an alternate copy of the CircuitPython firmware.

protosam, 2022-07-26:
Was thinking a copy of the circuitpython binary could be maintained to replace what's loaded. Though, that's problematic if the controller is in a product I guess.

I just learned that "run" is the equivalent of "reset" on the feather r2040 after looking at the code.

dansteingart, 2022-12-03:
Just re-upping this. `safemode.py` would be a welcome and IMHO crucial addition.

PR #7577 (2023-02-16): Implement safemode.py
Fixes #5956.
Fixes #2694.

- Add `safemode.py`, which is executed first, instead of `boot.py` and/or `code.py`, but only if the board has been reset due to going into safe mode.
- `safemode.py` has no access to USB. Its output is _not_ written to `boot_out.txt`, unlike `boot.py`, to minimize possible problems. There is a special safemode which is entered if there are errors in `safemode.py`.
- If `safemode.py` exits normally, the safe mode will remain in effect, and `boot.py` and `code.py` will not be executed. The safe mode reason will be reported in the console as usual [... cut]
closed 2023-02-16 by dhalbert