adafruit/circuitpython #5972: pull no block does not seem to work

DaveAtKitronik, 2022-02-02:


DaveAtKitronik, 2022-11-22:
Is there any news on this bug? It pervents us supporting Circuit Python on some of our products

tannewt, 2022-11-22:
Nope, it isn't high on our (Adafruit-funded) list. We're always happy to help others fix things though. The places to start are [the implementation](https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/common-hal/rp2pio/StateMachine.c#L838) and #circuitpython-dev on [our Discord](https://adafru.it/discord).

I'd also suggest comparing the full assembled versions between MP and CP. Its possible the bug is somewhere else besides the noblock.

jepler, 2022-11-28:
I wrote a program to check whether `pull noblock` worked to my understanding. It implements a glitch-free PWM with 16-bit high- and low times.

I'm on an Adafruit MacroPad but this should work on any RP2040 board with board.LED as a GPIO pin.

The LED alternates between "dim" (duty cycle 128 on : 16383 off) and "visibly flashing" (duty cycle 32767:32767)

This is my full code, running on CircuitPython 8 beta with current adafruit_pioasm:

```python
import array
import time

import board
from rp2pio import StateMachine
from adafruit_pioasm import Program

p = Program("""
    pull            ;; don't start until first value available
.wrap_target
    pull noblock
    mov x, osr      ;; reload OSR with last value

    set pins, 0
    out y, 16       ;; off time
loop_off:
    jmp y--, loop_off

    set pins, 1
    out y, 16       ;; on time
loop_on:
    jmp y--, loop_on

.wrap
""")

s = StateMachine(p.assembled, frequency=655_360, first_set_pin=board.LED, **p.pio_kwargs)

duty1 = memoryview(array.array('HH', [32767, 32767])).cast('L')
duty2 = memoryview(array.array('HH', [16383, 128])).cast('L')

while True:
    print(end="D")
    s.write(duty1)
    time.sleep(1)
    print(end="d")
    s.write(duty2)
    time.sleep(1)
```

Let's continue this elsewhere as agreed, and open a new issue with a reproducer if appropriate in the future.

closed 2022-11-28 by jepler