adafruit/circuitpython #5982: Countio on M4 Express doesn't work when  counter is imported

rp10007, 2022-02-03:
### CircuitPython version

```python
Adafruit CircuitPython 7.1.0 on 2021-12-28; Adafruit ItsyBitsy M4 Express with samd51g19
```

### Code/REPL

```python
code.py (finally working version):
from BrocItsyHelper import *

#encoder.deinit()
encoder = countio.Counter(board.MISO)

motor1.throttle=1; time.sleep(3); motor1.throttle = None
print(encoder.count)

BrocItsyHelper.py (works with above code.py, failed code commented out)
# BrocItsy helpers

import time
import board
import pulseio
import pwmio
import digitalio
import countio
#import adafruit_dotstar
import adafruit_hcsr04
from adafruit_motor import servo, motor

pwm = pwmio.PWMOut(board.D7, duty_cycle=0, frequency=50)
steering = servo.Servo(pwm)

#encoder = digitalio.DigitalInOut(board.D12)
#encoder.pull = digitalio.Pull.UP

#encoder = countio.Counter(board.D2)
#encoder.deinit()
#encoder = countio.Counter(board.MISO)
#encoder.deinit()
#encoder = countio.Counter(board.MISO)

#led = adafruit_dotstar.DotStar(board.APA102_SCK, board.APA102_MOSI, 1, brightness=0.2)
sonar = adafruit_hcsr04.HCSR04(trigger_pin=board.A3, echo_pin=board.A5)

print(22)

LEFT = 110
RIGHT = 180
MIDDLE = (LEFT + RIGHT) / 2

m1a = pwmio.PWMOut(board.D9)
m1b = pwmio.PWMOut(board.D10)
motor1 = motor.DCMotor(m1a, m1b)

def safesonar():
    try:
        return sonar.distance
    except:
        return 10000

def waitsonar(dist=10):
    while safesonar() < dist:
        time.sleep(2)

def forward(duration, speed):
    steering.angle = MIDDLE
    now = time.monotonic()
    motor1.throttle = speed
    while time.monotonic() < now + duration:
        if safesonar() < 10:
            motor1.throttle = None
            return "proximity"
        time.sleep(0.1)
    motor1.throttle = None
    return "finished"

def fwdcounter(duration, speed):
    steering.angle = MIDDLE
    now = time.monotonic()
    motor1.throttle = speed
    while time.monotonic() < now + duration:
        if safesonar() < 10:
            motor1.throttle = None
            return "proximity"
        time.sleep(0.1)
    motor1.throttle = None
    print (encoder.count)
    return "finished"

def backwards(duration, speed):
    steering.angle=MIDDLE
    motor1.throttle = -speed
    time.sleep(duration)
    motor1.throttle= None
    return "finished"

def left(duration, speed):
    steering.angle=LEFT
    now = time.monotonic()
    motor1.throttle=speed
    while time.monotonic() < now + duration:
        if safesonar() < 10:
            motor1.throttle = None
            return "proximity"
    return "finished"

def right(duration, speed):
    steering.angle=RIGHT
    now = time.monotonic()
    motor1.throttle=speed
    while time.monotonic() < now + duration:
        if safesonar() < 10:
            motor1.throttle = None
            return "proximity"
    return "finished"

def spin(direction, duration, speed):
    if direction in (left, counterclockwise, widdershins):
        steering.angle=RIGHT
        motor1.throttle=-speed
        time.sleep(duration/2.0)
        motor1.throttle=None
        result = left(duration/2, speed)
        if result == "finished":
            return "finished"
        else:
            return "proximity"
    else:
        steering.angle=LEFT
        motor1.throttle=-speed
        time.sleep(duration/2.0)
        motor1.throttle=None
        result = right(duration/2, speed)
        if result == "finished":
            return "finished"
        else:
            return "proximity"
def countchanges(duration):
    now = time.monotonic()
    oldval = encoder.value
    counter = 0
    motor1.throttle = 1
    while time.monotonic() - now < duration:
        if oldval != encoder.value:
            counter += 1
            oldval = encoder.value
            print (oldval)
    motor1.throttle = None
    print (counter)

def countchanges2(duration):
    now = time.monotonic()
    encoder.reset()
    motor1.throttle = 1
    time.sleep(duration)
    motor1.throttle = None
    print (encoder.count)
```

### Behavior

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
22
0
(that zero is the result of the counter after running motor when I define it in the helper file)

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
22
16
(that's what happens when I define the counter in the code.py file)

### Description

On the M4 Express (but not afaict on the pi2040) Itsy Bitsy,  using countio to define a counter in a helper file, and then doing "_from helper import *_" in the code.py file or in the repl leads to a nonworking counter object. The object identifies itself as <Counter>, there's an error message if I try to define another counter on the same pin, but the count remains stubbornly at 0. (I checked the hardware side by doing deinit, creating a Digitalio object and verifying that it alternates between true and false.)

If I deinit the counter object in the repl and create a new one on the same pin, then it works. Deinit and recreation in the helper file: nope. If I leave the counter object out of the helper file and create it in code.py, it works. 

### Additional information

This can be addressed by just doing the creation of counters in code.py on the M4. (But in addition to offending one's sense of elegance, I think it might be a sign of something wonky in a deeper way.) I also just double-checked my code from a 2040 version, and defining counters in a helper file appears to work fine there. (The 2040 does extensive checking on which pins are OK for counters, which the M4 express appears not to do.)

Thanks!

tannewt, 2022-02-04:
This feels like a bug due to long living the Counter object. We must have some memory internally that we move but shouldn't.

dhalbert, 2022-02-08:
A pointer to the object is used as the EIC channel data:
```c
    set_eic_channel_data(self->eic_channel, (void *)self);
```
If the object moves, this will become invalid (and interrupt routine will start incrementing in a non-Counter object). @tannewt, I'm not sure what you mean about the long-living. When would it be set to long-lived?

tannewt, 2022-02-08:
Sorry, I meant this was due to us *not* long living the object earlier. The module's contents will be long lived once it finishes importing and will move the pointer given in that function.

PR #6106 (2022-03-02): Make countio object long-lived
Fixes #5982.  Made the countio object long-lived so it persists across imports.
closed 2022-03-02 by dhalbert