adafruit/circuitpython #5986: Support USB host (rp2)

dhalbert, 2022-02-06:
From @ATMakersBill in https://github.com/adafruit/circuitpython/issues/5630#issuecomment-1030939071:

>USB Host support (on any board) would be hugely helpful to our community. I've seen this solution https://github.com/sekigon-gonnoc/Pico-PIO-USB and would love to see something like that make it into CP (even if only on the RP2040) at some point. Either way, thanks for working on this.

ATMakersBill, 2022-02-06:
Thanks for adding this, @dhalbert!

We have a number of "conversion" projects that take old equipment and convert them to current standards.  The most successful is the IntelliKeys adapter https://github.com/ATMakersOrg/IKeysAdapter  It takes an old product that is STILL the best device for many folks with cerebral palsy and makes it work with any modern operating system.

The solution we made took two Trinket M0's back to back and talked over UART.  It works, and we've got over a hundred in the field now.  But it's clunky, costs more than it should & relies on weird cables (micro-B to Full-Sized B).  @ladyada was good enough to add the diode bypass on the RP2040 QT Py which lets us use those, but the host library we're working doesn't support the RP2040 yet, so we're using a weird hybrid: CircuitPython on the RP2040 talking UART to a Trinket M0 running Arduino.  It works, but again, it seems clumsy.

I recently saw this solution that uses the RP2040's PIO system to add a USB Host port on two GPIO pins https://github.com/sekigon-gonnoc/Pico-PIO-USB.  It supports everything we'd need, but is not in a useful environment (neither Arduino nor CP).  If this same PIO code was to work as a CP module (or arduino library), it would be amazing.  I can even see simple #BFF or #FeatherWing boards that just had a female USB-A jack and some passives.

Anyway, please know that we'd put this to very good use - let me know if I can help
Bill

dhalbert, 2022-04-11:
@tannewt is off for 4-5 weeks, so I'd say to wait. He will be back and be able to answer questions authoritatively then.

tannewt, 2022-05-09:
@rycwo I don't believe TinyUSB has support in its main branch yet. It's waiting on https://github.com/sekigon-gonnoc/Pico-PIO-USB/pull/22

So, progress is being made but isn't there yet. Once it is, we'll also need to connect the actual reading and writing of packets to the new TinyUSB API for it. You could start this on iMX for now if you are eager. We're happy to help ramp folks up to contributing to CircuitPython. Chatting with us on Discord is a good place to start: https:/adafru.it/discord

Thanks!

rycwo, 2022-05-11:
Thanks for the info. I'll reach out on Discord for more information!

closed 2023-07-11 by tannewt

joshmarinacci, 2023-07-11:
Wow, this is huge!  Thank you!

tannewt, 2023-07-11:
> Wow, this is huge! Thank you!

Thanks! It isn't bug free so please file issues as you find them.

spencerthayer, 2023-07-12:
I *just* saw your [YouTube live video](https://www.youtube.com/live/jJmAtxn2LKA)  @tannewt. Amazing timing that you just completed it today! Congrats!

Anyway in that video you casually mentioned eventually being able to connect a USB device to an ESP32-S3 Feather and use it with an iPad over Bluetooth. I am looking into possibly doing that for my ZSA Moonlander, any ideas on how one would accomplish this feat using circuitPython?

chabala, 2023-07-13:
> We don't have a device that does usb host and ble unfortunately.

Wouldn't that just be any of the Pico form factor RP2040 boards + https://www.waveshare.com/wiki/Pico-BLE ?

PR #8155 (2023-07-11): Basic USB host support and keyboard workflow
Connects up read, write and ctrl_transfer to TinyUSB. USB Host support is available on iMX RT and RP2040.

Fixes #6527 (imx) and fixes #5986 (rp2).
commit 2686beab (2023-07-11): Basic USB host support and keyboard workflow

Connects up read, write and ctrl_transfer to TinyUSB. USB Host
support is available on iMX RT and RP2040.

Fixes #6527 (imx) and fixes #5986 (rp2).