adafruit/circuitpython #5992: Build failure for raspberry PI 4

waqas-shafique, 2022-02-08:
### CircuitPython version

```python
circuitpython-7.2.0-alpha.1
```

### Code/REPL

```python
N/A
```

### Behavior

make BOARD=raspberrypi_pi4b
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
ERROR: Empty preprocessor output - check for errors above
make: *** [../../py/py.mk:269: build-raspberrypi_pi4b/genhdr/qstrdefs.enum.h] Error 1
make: *** Deleting file 'build-raspberrypi_pi4b/genhdr/qstrdefs.enum.h'

jerryneedell, 2022-02-08:
It builds OK for me -- did you install everything as in  this guide <https://learn.adafruit.com/building-circuitpython>?

waqas-shafique, 2022-02-08:
I tried all steps again with a fresh clone of the repo but the same result as before. 

I tried to build another board using make BOARD=circuitplayground_express and it builds fine. 

However, for make BOARD=raspberrypi_pi4b  I'm getting the same error i.e. 

GEN build-raspberrypi_pi4b/genhdr/mpversion.h
python3 ../../py/makeversionhdr.py build-raspberrypi_pi4b/genhdr/mpversion.h
GEN build-raspberrypi_pi4b/genhdr/qstrdefs.enum.h
python3 ../../py/makeqstrdata.py build-raspberrypi_pi4b/genhdr/qstrdefs.preprocessed.h > build-raspberrypi_pi4b/genhdr/qstrdefs.enum.h
ERROR: Empty preprocessor output - check for errors above
make: *** [../../py/py.mk:269: build-raspberrypi_pi4b/genhdr/qstrdefs.enum.h] Error 1
make: *** Deleting file 'build-raspberrypi_pi4b/genhdr/qstrdefs.enum.h'

Also git describe --tags command shows below result
7.2.0-alpha.1-343-gfd2cbc00f

closed 2022-02-08 by tannewt