adafruit/circuitpython #6005: audiopwmio freezes/crashes CircuitPython sometimes with MSD access

todbot, 2022-02-09:
### CircuitPython version

```python
Adafruit CircuitPython 7.1.1 on 2022-01-14; Adafruit ItsyBitsy RP2040 with rp2040
```

### Code/REPL

```python
import time
import board
import audiocore
import audiomixer
from audiopwmio import PWMAudioOut as AudioOut  # for RP2040 etc
#from audioio import AudioOut as AudioOut  # for SAMD51 etc

# audio pin is almost any pin on RP2040, let's do RX (pin 1) (RP2040 GPIO1) or A0 (RP2040 GPIO26)
# audio pin is A0 on SAMD51 (Trelllis M4, Itsy M4, etc)
audio = AudioOut(board.A0)
mixer = audiomixer.Mixer(voice_count=1, sample_rate=22050, channel_count=1,
                         bits_per_sample=16, samples_signed=True)
audio.play(mixer) # attach mixer to audio playback

print("audio is now playing")

# start drum loop playing
wave0 = audiocore.WaveFile(open("wav/drumsacuff_22k_s16.wav","rb"))
mixer.voice[0].play( wave0, loop=True )

while True:
    time.sleep(0.1)
```

### Behavior

When an RP2040-based device running the above `audiopwmio` code is plugged into a USB host or USB MSD activity (saving `code.py`), extremely loud audio glitches result as the host does various USB activity (presumably mostly MSD).  Sometimes CircuitPython will freeze and crash entirely with the REPL message:
```
Code stopped by auto-reload.
```

### Description

Sometimes, the audio glitching causes all of CircuitPython to freeze: REPL dead, cannot access CIRCUITPY drive, etc.  In those cases, the host will eventually register a disconnect from the device.

The attached video shows what I'm seeing: the glitches can be heard, CircuitPython locks up, and glitches are heard during `code.py` saving. 

Warning: loud sounds 

https://user-images.githubusercontent.com/274093/153290112-28180f6a-af25-471b-8fb8-aba473b3e6e7.mp4

### Additional information

This has been experienced on ItsyBitsy RP2040, KB2040, and Raspberry Pi Pico. 

On `audioio` boards like SAMD51 ItsyBitsy M4 with DAC, instead of `audiopwmio`, there is minimal to no glitching.

Suggestion: maybe MSD access can pause `audiopwmio` (if active) for 500 milliseconds?  This would reduce most glitches and maybe "get rid" of source of crash by eliminating DMA/interrupt contention?

todbot, 2022-02-09:
Correction: sometimes saving `code.py` on a SAMD51 M4 board (Trellis M4 in this case) running `7.1.1 on 2022-01-14` will also freeze CircuitPython.

tannewt, 2022-02-10:
Do you have a GDB setup? Connecting over GDB and getting a backtrace is a great way to debug a freeze.

I imagine RP2040 is tricky because it's loading code, USB data and audio data all from the flash.

todbot, 2022-02-10:
I'll look into getting GDB set up for RP2040. I assume someone's getting their interrupt missed.

dhalbert, 2022-02-11:
I have not had a lot of success getting gdb to work well with the RP2040. I have tried mostly with the J-Link.

gallaugher, 2022-02-20:
Similar problems with similar code. Tried on two RP2040 boards - Ardino Nana RP2040 Connect and QT Py RP2040 with same problematic results. Tried code on a CPB and it worked without these probs. 
Have been using RP2040s in lots of situations but this is the first I’ve had these consistent problems. Also the first time I’ve used audiocore and audiomixer. Earlier work with PWMAudioOut didn’t seem to run into these problems (about 50 students, two semesters, but didn’t hear regular complaints)
My code is based off @todbot’s and is here:
https://github.com/gallaugher/breakbeats-mpr121
Thx!

gallaugher, 2022-02-20:
Correction.
- Now I'm experiencing the same problem with CircuitPlayground Bluefruit. I can enter Safe Mode, but can't re-enter standard mode & run my code on the CPB, just like the RP2040 models.
- I had been using audiocore previously on both boards without any problems (just importing WaveFile. The only thing that's new is audiomixer, so perhaps that's the problem.

jedgarpark, 2022-07-01:
@dhalbert @tannewt  I still have problems doing any coding on an RP2040 that plays audio because of this issue. Iteration is difficult because any time I want to save code.py to have it restart and test changes I need to carefully time pressing reset button on board, getting into REPL and hitting ctrl-c first before code starts running, then save my code and hit ctrl-d.

todbot, 2022-11-16:
As I've played with this more, adding `audiomixer` definitely causes more problems. 

and btw, WAV file used in my video above is at https://github.com/todbot/circuitpython-tricks/blob/main/larger-tricks/wav/drumsacuff_22k_s16.wav

jepler, 2022-11-29:
I tried to reproduce this in `Adafruit CircuitPython 7.3.3 on 2022-08-29; Adafruit QT Py RP2040 with rp2040` using @todbot's original reproducer script but failed.  

To simulate interaction with the CIRCUITPY filesystem I ran a script which writes 16kB of random data and then waits 4 seconds before doing it again. I let it run for dozens of iterations.

When I flashed back to 7.1.0, the reproducer _DID_ work, corrupting CIRCUTIPY.

I suspect that the re-worked auto-reload in 7.2.x probably played a part in fixing this.

I didn't actually listen to the audio, since this issue is about crashing. Audio playback of wav files during flash writing will not work properly, because the wave data can't be read during a flash erase or flash write operation.

Closing as I verified the reproducer no longer causes CIRCUITPY to become corrupted.

closed 2022-11-29 by jepler

todbot, 2022-11-29:
I slightly disagree with closing this, but okay.  I had only rarely observed audio causing a corruption of CIRCUITPY.

My main concern for this issue was two-fold:

1. Playing audio with audiomixer sometimes crashes the supervisor (but doesn't corrupt CIRCUITPY, as seen in my video above)
2. The audio artifacts are disruptive to the point of potentially damaging to downstream audio systems

jepler, 2022-11-29:
How long do you recommend I run your code.py script + my CIRCUITPY writing / reloading code before I conclude that there is no CircuitPython crash or CIRCUITPY corruption happening anymore?  100 reloads?  1000 reloads?

todbot, 2022-11-29:
In my experience (as seen in the above video), simply resetting the board shows the problem immediately.

todbot, 2022-11-29:
I just tested this on a QTPy RP2040 running `CircuitPython 8.0.0-beta.4-55-gd364d1c51`, using `audiopwmio` into `audiomixer`, and the results are much improved!  See attached video.  (`audiopwmio` seemed to be the worse offender, compared to `audioio` and `audiobusio`)

There is still the very disruptive audio glitching that happens if the device is hooked to a host OS (presumably as TinyUSB MSD uses up the cycles as CIRCUITPY is mounted).   But I have not been able to get it to hard crash just from pressing RESET or Ctrl-D.  Before there was about a 90% chance it would crash.  

So I am okay with this being closed.  This seems to no longer be a frustrating work-stopper like before, where we had to stick `time.sleep(5)` at the top of our code and jump on Ctrl-D as soon as CDC connected to prevent the inevitable crash.

Thanks!

https://user-images.githubusercontent.com/274093/204646498-518a5c76-2e97-4a73-a649-acb85274c380.mp4

jepler, 2022-11-29:
In rp2040, there are two DMA channels used.  First, DMA channel 1 sends a batch of samples, then it triggers (chains to) channel 2.  Channel 2, on completion, chains back to channel 1.

The DMA completion signal also sets a bit in the value called "dma->channels_to_load_mask", which is later checked by `dma_callback_fun`. When bits in this value are set, it copies fresh audio data into one of the buffers.

I think the problem is that the audio DMA comes from RAM, so it can continue while there's a flash erase/write in progress; but the update code in `dma_callback_fun` and `isr_dma_0` can't be run. So, old audio data continues looping until the flash erase/write cycle completes.

If the dma channels did not "chain", but relied on `isr_dma_0` to start the other channel only if fresh data is available in it, this might replace the stuttering effect with a pop followed by silence, followed by another pop and audio when the buffer is re-filled. It's not clear if this would be better.

tannewt, 2022-11-30:
We could place the audio buffer fill functions in RAM but that would only help if the samples and any code to generate them were in ram as well.