adafruit/circuitpython #6007: No autoreload when saving a file on SAMD21 sometimes

Neradoc, 2022-02-10:
Since around 7.0 I have noticed on SAMD21 boards that sometimes the autoreload doesn't not happen.
It is hard to reproduce and seems to happen randomly. But once the boards stops running autoreload:
- saving files works, but does not trigger autoreload
- ctrl-C ctrl-D works, but does not bring back autoreload
Only a hard reset seems to bring it back.

I have seen the board enter that bad state:
- as soon as I plugged it in.
- after doing some file operations from the host side.
- after leaving the board plugged in without using it for an hour.

I have done some of those tests on a freshly-flashed, latest-main, and erased-file-system CircuitPlayground Express with a simple neopixel animation code.

DavePutz, 2022-02-17:
I reproduced this issue running CP 7.2.0-alpha.2. No autoreload occurred after editing code.py
The stack from gdb and the variables you mentioned:

```
#0  0x0000a628 in common_hal_mcu_enable_interrupts ()
#1  0x0001ccf8 in port_idle_until_interrupt ()
#2  0x00015a36 in mp_hal_delay_ms ()
#3  0x000237cc in time_sleep ()
#4  0x000100ee in fun_builtin_1_call ()
#5  0x000140a0 in mp_call_function_n_kw ()
#6  0x000299e6 in mp_call_method_n_kw ()
#7  0x00015070 in mp_execute_bytecode ()
#8  0x00010078 in fun_bc_call ()
#9  0x000140a0 in mp_call_function_n_kw ()
#10 0x000299b6 in mp_call_function_0 ()
#11 0x0001c5b6 in parse_compile_execute ()
#12 0x00012548 in maybe_run_list ()
#13 0x000201a0 in run_code_py.constprop.0.isra ()
#14 0x00021522 in main ()

(gdb) x /d 0x000000002000134c
0x2000134c <autoreload_delay_ms>:       500
(gdb) x /d 0x0000000020001350
0x20001350 <autoreload_suspended>:      0
(gdb) x /x 0x0000000020001cac
0x20001cac <autoreload_enabled>:        0x00000c01
```

jerryneedell, 2022-02-18:
Just another observation. This happened to me on a feather_m0_adalogger running 7.1.1. 
Auto reload stopped. I tried copying code.py several times. No auto reload, but on control_d the new file ran.
The issue was resolved by a hard reset.
I don't have any other details, sorry.

tannewt, 2022-02-18:
Thanks for reproducing this @DavePutz! Nothing looks out of ordinary there. Maybe it isn't ticking and counting down?

DavePutz, 2022-02-21:
Yes, when autoreload stops working autoreload_delay_ms is always stuck at 500. So, it's not running autoreload_tick(); which means supervisor_tick() isn't getting called from evsyshandler_common(). I am not sure how the interaction between tinyusb and the event subsystem is supposed to work, however.

tannewt, 2022-02-22:
I think tinyusb should just start the tick using `supervisor_enable_tick` via `autoreload_start()`. It is interesting that there is a check that autoreload_delay_ms == 0 to start the tick. Here it is 500, so it must have tried and failed to start the tick. Or ticks were disabled separately while autoreload_delay_ms was non-zero.

ladyada, 2022-03-02:
for me it was a prox trinkey, using the repl and doing some HID development. i plugged it back in and now auto-reload is working :/

FoamyGuy, 2022-03-02:
I have an Itsy Bitsy M0 `7.2.0` that I noticed this issue on the other day. I am able to reproduce it reliably with every code change currently using a basic code.py that prints different things. I can update `code.py` with new print messages and then save, IDE reports file saved (astrisk disappears) but code is not rerun. I can either reset button, or ctrl+c then ctrl+d to run the latest code.

FoamyGuy, 2022-03-02:
The issue does seem to be tied to the REPL. I disconnected `tio` and subsequent changes to code.py do get executed when the file is saved.

When the REPL is connected via `tio` then it will consistently not update upon saving code.py. It seems to me like it is not sporadic. Not running any time that I save. When the REPL is open I must use ctrl+C then ctrl+D to get the latest code to run at all.

tannewt, 2022-03-02:
Do you mean serial? When the REPL is active (with >>> prompt) auto-reload is intentionally disabled.

FoamyGuy, 2022-03-02:
Ah, yeah I did mean serial when connected via `tio` not specifically in the REPL.

dhalbert, 2022-03-02:
I also meant serial, not specifically REPL.

PR #6132 (2022-03-09): Make autoreload checking more robust
- Fixes #6007.

- Add reset for autoreload. De-request ticks.
- Separate state a little more in autoreload.c
- Rename some routines and variables.
- Reset `_tick_event_channel` when ticks are disabled, in atmel_samd/supervisor/port.c
- Add `PYEXEC_RELOAD` as a specific reason for a forced termination of code, instead of checking the exception.
- Remove redundant settings of `CIRCUITPY_AUTORELOAD_DELAY_MS`.

The most vital changes are the autoreload reset, and perhaps the `_tick_event_channel`.

Saving a file over and over again very quickly can still occasi [... cut]