adafruit/circuitpython #6009: Broadcom port on Zero W: unable to Pull.UP inputs

wmlele, 2022-02-10:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0-alpha.1-364-g2bf5a1ee4 on 2022-02-09; Raspberry Pi Zero W with bcm2835
```

### Code/REPL

```python
import board
from digitalio import DigitalInOut, Direction, Pull
import time

# switch
switch = DigitalInOut(board.D19)

# if I don't do this, it won't pull the input up 
#switch.direction = Direction.OUTPUT
#switch.value = True

switch.direction = Direction.INPUT
switch.pull = Pull.UP

while True:
    print(switch.value)
    time.sleep(0.5)
```

### Behavior

Level on D19 stays at 0V when open regardless of Pull.UP. `switch.value` stays `False` regardless of the state of the input. 

If the level is first set as _output_, and manually set to a **true** value, then it stays up after reverting to _input_, and I can read the state.

tannewt, 2022-02-10:
Do you have any experience with C on the Pi? The code for setting pulls is here: https://github.com/adafruit/broadcom-peripherals/blob/08370086080759ed54ac1136d62d2ad24c6fa267/broadcom/gpio.c#L24

Called from https://github.com/adafruit/circuitpython/blob/main/ports/broadcom/common-hal/digitalio/DigitalInOut.c#L137

I'd love help polishing up this port.

Florin9doi, 2023-09-10:
Possible fix: https://github.com/Florin9doi/broadcom-peripherals/commit/44f5392759c572502e3707e95ab2dfd68fb184da

cleverca22, 2023-09-10:
i think that commit is about bcm2711 pulls, which differ
but the zero-w, is using the vc4 pull style, which is write-only