adafruit/circuitpython #6012: scanning while advertising causes NRF_ERROR_BUSY: _bleio.BluetoothError: Unknown system firmware error: 0011

awgrover, 2022-02-10:
Several examples are available that show scanning while advertising (i.e. no stop_advertising). E.g. [Clue rock-paper-scissors](https://learn.adafruit.com/rock-paper-scissors-circuitpython/simple-game), and some non circuitpython examples for the nRF52840 (e.g. https://learn.adafruit.com/introducing-the-adafruit-nrf52840-feather/dual-roles-bleuart).

On the bluefruit at `.start_scan()`, I get "_bleio.BluetoothError: Unknown system firmware error: 0011", which is NRF_ERROR_BUSY, implying scanning can't occur during advertising. I can find no other report of this error (which seems odd). Code works under circuitpython 6.3.0.

This appears to be some change in _bleio in 7.x.x (since 6.3.0).
 
Hardware, as reported in boot_out.txt: 
Adafruit Circuit Playground Bluefruit with nRF52840
Board ID:circuitplayground_bluefruit

circuitpython, as reported in boot_out.txt and repl:
Adafruit CircuitPython 7.1.1 on 2022-01-14; Adafruit Circuit Playground Bluefruit with nRF52840

only 1 library in lib/ folder:
adafruit-circuitpython-ble-7.x-mpy-8.2.0/lib/adafruit_ble
reported by repl:
>>> import adafruit_ble
>>> print( adafruit_ble.__version__ )
8.2.0

(I tried the "simple game" of Clue rock-paper-scissors, same error.)
Circuitpython vs adafruit_ble matrix: 

                            adafruit_ble
                            8.2.0   8.1.0   7.3.4
    circuitpython
    7.1.1                   fail    fail    fail
    en_US-20220210-c234d92  fail            fail
    6.3.0                   works           works

**example code**
```
import traceback # remove for circuitpython 6.3.0
import time

import _bleio # for the exception
from adafruit_ble import BLERadio
from adafruit_ble.advertising import Advertisement

print("\nstart")

MAX_SEND_TIME_S = 5

# 20ms is the minimum delay between advertising packets
# in Bluetooth Low Energy
# extra 10us deals with API floating point rounding issues
MIN_AD_INTERVAL = 0.02001 # changing has no effect

ble = BLERadio()

advertisement = Advertisement()
advertisement.short_name = "HELLO"
advertisement.connectable = False # True has no effect

print("loop")

# related to the previous bonding? no
# ble._adapter.erase_bonding()

# advertise while scanning:
ble.start_advertising(advertisement, interval= MIN_AD_INTERVAL)

while True:
    # does waiting have any effect? no
    # time.sleep(0.2)

    # Timeout value is in seconds
    # window and interval are 0.1 by default - same value means
    # continuous scanning (sending Advertisement will interrupt this)
    gotit = 10
    while(gotit > 0):
        try:
            # the .start_scan() will throw:
            # Traceback (most recent call last):
            #   File "code.py", line 47, in <module>
            #   File "adafruit_ble/__init__.py", line 250, in start_scan
            # _bleio.BluetoothError: Unknown system firmware error: 0011

            for adv in ble.start_scan(
                                      Advertisement,
                                      # does non-continuous have any effect? no
                                      #window=0.1,
                                      #interval=0.2,
                                      #minimum_rssi=-90,
                                      # does a buffer have any effect? no
                                      #buffer=256,
                                      # does a timeout have any effect? no
                                      # timeout=MAX_SEND_TIME_S
                                      ):
                gotit = 0
                print("received: ", adv.short_name)

                ble.stop_scan()

                break
        except _bleio.BluetoothError as e:
            # does retrying have any effect? no

            # should we have done a stop_scan for retry? no
            #ble.stop_scan()
            gotit -= 1
            time.sleep(0.03)
            if (gotit==9): # we don't need to see it 10 times
                print( traceback.format_exception( None, e, e.__traceback__) )
            print("fail...",gotit, e) # fails 0..9
    if gotit == 0:
        print("Failed")
        break
```

tannewt, 2022-02-11:
I think this is due to 7.0 loading the peer keys for resolving addresses. IIRC the soft device only has one key store and we load different keys depending on whether we are a central or peripheral. Maybe we can load both sets at once to allow concurrent scanning and advertising.

@awgrover Do you have devices bonded to you when you are doing both scanning and advertising?

awgrover, 2022-02-13:
@tannewt No devices are bonded. I tried `ble._adapter.erase_bonding()` which had no effect.

tannewt, 2022-02-14:
Thanks! I'll try to look into this in the next few weeks.

dhalbert, 2022-03-22:
Could have key store load everything, or check that there is no bonding.

kevinjwalters, 2022-04-16:
I've just noticed this on `7.2.0`, I tried break on a `start_scan()` for loop and got this, thought I could work around it by waiting for the timeout to drop out of the for loop but that doesn't work either.

kevinjwalters, 2022-04-16:
I think it does it all the time:

```
Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.2.0 on 2022-02-24; Adafruit CLUE nRF52840 Express with nRF52840
>>>
>>> from adafruit_ble import BLERadio
>>> from adafruit_ble.advertising.adafruit import AdafruitColor
>>> ca = AdafruitColor()
>>> ca.color = 0x778899
>>> radio = BLERadio()
>>> for adv in radio.start_scan(timeout=3.0):
...     pass
...
...
...
>>> radio.start_advertising(ca)
>>> for adv in radio.start_scan(timeout=3.0):
...     pass
...
...
...
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_ble/__init__.py", line 250, in start_scan
_bleio.BluetoothError: Unknown system firmware error: 0011
```

kevinjwalters, 2022-04-16:
BTW, I think this will break any sub-classes of `Advertisement` too as they won't have an entry `kwarg` on their constructor, etc.

https://github.com/adafruit/Adafruit_CircuitPython_BLE/commit/b6810c8e0a9d7a38e397d23b1b24f1378c74629d#diff-c0cf323ca9b06c937667387c8f6d305e72349e6a0ad00b665267bb13ae6f6234R239

cHemingway, 2022-05-16:
I am also experiencing this problem on CircuitPython 7.2.5 on the nRF52840. 
If I stop advertising while scanning, it works.

So this (pseudocode) works, in a loop

```python
while True:
    ble.stop_advertising()
    ble.start_scan(...)
    # do whatever with the scan results
    ble.stop_scan()
    ble.start_advertising(...) 
```

burksfamly, 2024-02-20:
I confirmed cHemingway's findings for the CLUE Morse Code Chat example.
BL 0.8.0 and 0.8.2
CP 8.x and CP 9.x beta
CP Lib 8.x and CP 9.x beta
Morse Code Example Bundle 8.x and 9.x

The only way to get this example to work is to modify the scan_and_connect function to perform scanning if the 
CLUE is to be the central and to perform advertising if the CLUE is to be the peripheral.

`MY_NAME = "CENTRAL"
FRIENDS_NAME = "PERIPHERAL"

#MY_NAME = "PERIPHERAL"
#FRIENDS_NAME = "CENTRAL"

def scan_and_connect():
    '''
    Advertise self while scanning for friend. If friend is found, can
    connect by pressing A+B buttons. If friend connects first, then
    just stop.

    Return is a UART object that can be used for read/write.
    '''

    # peripheral
    if MY_NAME == "PERIPHERAL":
        print("Advertising.")
        central = False
        ble.start_advertising(advertisement)

        while not ble.connected:
            if ble.connected:
                break

        # We're now connected, one way or the other
        print("Stopping advertising.")
        ble.stop_advertising()

        print("Peripheral - connecting to their UART service.")
        for connection in ble.connections:
            if UARTService not in connection:
                continue
            return connection[UARTService]

    # central
    else:
        print("Waiting.")
        friend = None
        while not ble.connected:
            if friend is None:
                print("Scanning.")
                in_label.text = out_label.text = "Scanning..."
                for adv in ble.start_scan():
                    if ble.connected:
                        # Friend connected with us, we're done
                        ble.stop_scan()
                        break
                    if adv.complete_name == FRIENDS_NAME:
                        # Found friend, can stop scanning
                        ble.stop_scan()
                        friend = adv
                        print("Found", friend.complete_name)
                        in_label.text = "Found {}".format(friend.complete_name)
                        out_label.text = "A+B to connect"
                        break
            else:
                if clue.button_a and clue.button_b:
                    # Connect to friend
                    print("Connecting to", friend.complete_name)
                    ble.connect(friend)
                    central = True
        

        # Return a UART object to use
        print("Central - using my UART service.")
        return uart_service`