adafruit/circuitpython #6019: Add `wait_for_fifo_empty` kwarg to PIO write

tannewt, 2022-02-11:
This would allow for loading data into the state machine using `run()` before the state machine is actually clocked. We can't do it now because `write()` blocks until the FIFO is empty.

When this is False, `write()` needs to ensure that the length of the data is less than or equal to the FIFO size because it won't be able to queue more data up later.

YashasviChaurasia, 2022-08-09:
@tannewt I would like to work on this issue.
Thank you

tannewt, 2022-08-09:
@YashasviChaurasia A PR is welcome! Join us on Discord for help with circuitpython development. #circuitpython-dev on https://adafru.it/discord

deshipu, 2023-07-29:
@YashasviChaurasia are you still working on this, or is it free ot be taken by other people?

YashasviChaurasia, 2023-08-03:
Sorry for the delay, This issue can be taken up by others.
thank you